open62541/open62541 #1063: When unit tests dont compile, travis is all green.

Infinity95, 2017-06-02:
Description
===========

As the title says, travis is green, when the unit tests don't compile.
This can bee seen in PR #1003. This also happens in my [local repository on a pull request](https://travis-ci.org/Infinity95/open62541/builds/238804478?utm_source=github_status&utm_medium=notification), and also in my local branch.
I'd assume theres something wrong with the travis configuration maybe?

Background Information / Reproduction Steps
===========================================

Make a unit test not compile. Travis will (probably) stay green.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 6b3186ab5d6750f9ec6b2451b631d6b95abb7c63
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-06-02:
Interesting...
I'll have a look at it in the following days. It definitely worked before that failed unit tests also caused failed travis builds.

Infinity95, 2017-06-02:
Cool. Tell me if you need anything else concerning this.

Pro, 2017-06-07:
Fixed in #1066

PR #1066 (2017-06-07): Fix travis not erroing on build fail
See https://github.com/travis-ci/travis-ci/issues/1066

Fix #1063
closed 2017-06-07 by Pro