open62541/open62541 #1069: New ReferenceNodeKind structure not supported by type system (Init/Delete) and leads to SEGVs

janitza-thbe, 2017-06-08:
Description
===========
With commit 1f7e07d0 from 31.05.2017 Julius Pfrommer introduced the new structure ReferenceNodeKind and used it to handle all references for nodes.
Unfortunalety the required code to automatically and recursively delete/deleteMembers was neither added nor manually added to a custom types description file (bsd suffix) to generate these instructions automatically as I analysed so far.

Currently each call of UA_delete() or UA_Array_delete() is not possible or invalid, e.g. in ua_services_nodemanagement.c:379. There is set still the illegal old reference structure type.

`UA_Array_delete(node->references, node->referencesSize, &UA_TYPES[UA_TYPES_REFERENCENODE]);`

A further call will lead immediately to SEGVs because the algorithm tries to free() illegal memory addresses. Especially with multi-threading enabled where this instruction is called so often for nearly every node change/copy the stack is completely unusable currently.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:
 - [x] open62541 Version (release number or git tag): 1f7e07d0
 - [x] Critical issue

jpfr, 2017-06-08:
Hello there,

of course, You cannot expect to delete a ReferenceNodeKind structure with code that expects a UA_ReferenceNode structure.

Users are not expected to touch the node structure yourself. It is not exposed in the public API.
The ReferenceNodeKind structure is only used within a node.
And nodes are deleted via this function:
https://github.com/open62541/open62541/blob/master/src/server/ua_nodes.c#L8

Can you tell us about your use case? Then we can extend the public API in a sensible way.
We don't expose the node objects for a reason!
That way, we can improve the internal code without breaking user code.

jpfr, 2017-06-08:
Right. So it is actually a problem in the main library.
UA_Array_delete takes a void pointer. So this wasn't caught by the compiler.
I am surprised it didn't get caught by Travis.

The fix is forthcoming.

jpfr, 2017-06-08:
Fixed via #1070.

Thanks for the complete issue report.

commit 55f2cfab (2017-06-08): fix #1069 and add a test case for object type instantiation
closed 2017-06-08 by jpfr