open62541/open62541 #1071: After timeout of session, get crashed

JinGuoGe, 2017-06-08:
Description
===========
I am using source #1070 aa717f20e2. 
I am using open62541 for opcserver in Ubuntu 15.04 and in other side, using eclipse milo for opcclient in Windows.

Problem is after end of session activation, then open62541 based opcserver get crashed.

Background Information / Reproduction Steps
===========================================

While monitoring of session between opcclient and opcserver, i plugged off opcserver running pc's network.
In that case after 1~2 min (i think at end of session time), then get crashed.
Please refer crashed log  here: https://paste2.org/t9LpULgU

(I also tested 42187bba03	version too, and problem is more complex in that case, anyway please  help fixing about unpextected connection  between opcserver and opcclient while keeping monitoring session_
Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): aa717f20e2
 - [ ] Other OPC UA SDKs used (client or server):  eclipse milo
 - [ ] Operating system: ubuntu15.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

JinGuoGe, 2017-06-10:
Hi,  thanks for trying to solve now.
I am going to add valgrind log by hoping that this can help to find reason.
https://paste2.org/LyIC0VFd

I am anyalzing problem, As result, i think reason seems that after removing jobs by removeRepeatedJob() function,  processRepeatedJobs() is called at next iteration of running loop.  (UA_Server_run_iterate() function)

jpfr, 2017-06-15:
According to the valgrind log, the problem is sampling of values for a session that no longer exists.

First, this happens in removeRepeatedJob:
```c
    LIST_FOREACH_SAFE(rj, &server->repeatedJobs, next, rj_tmp) {
        if(!UA_Guid_equal(jobId, &rj->id))
            continue;
        LIST_REMOVE(rj, next);
        UA_free(rj);
        break;
}
```

Afterwards, there is a fault access in processRepeatedJobs:
``c
  LIST_FOREACH_SAFE(rj, &server->repeatedJobs, next, tmp_rj) {
        if(rj->nextTime > current)
            break;

         // process the job
    }
```

Something is odd about that.
You can see for yourself, that the pointer to the repeated job is no longer visible after removeRepeatedJob. So how can it be accessed?

You had some issues with multithreading before. Are you sure there is only on thread that calls UA_Server_run_iterate?

JinGuoGe, 2017-06-15:
There is only one thread that calls UA_Server_run_iterate. This problem is occurred when requested with multiple subscriptions, but when requested with one subscription (all monitoring request are from one subscription), then this problem was not occured.

jpfr, 2017-06-27:
Hello @JinGuoGe ,

we have to reproduce the error case to fix the issue and prevent future regressions.
For this, I added support for unit tests with simulated (precise) timings in #1089 .

If you have a dump of the network traffic (see the Wireshark tool) with the crash happening, writing the unit test can be done easily. Otherwise, it will take more time to reproduce the issue.

jpfr, 2017-07-04:
Hello there,

I rigged a setup where the virtual machine is killing it's network driver to simulate removing a network cable. (`ifdown / ifup` on linux). I let the securechannel and session time out, activate the network driver and reconnect the client.

This runs in a loop since a day now. So far, I did not see the crash you reported.
The process has valgrind and a debugger attached. So it stops at the first error.

Every information is helpful.
- Which client did you use. How many sessions and subscriptions did you run with which settings?
- If you see the crash again, can you provide a dump of the network traffic? See here: https://github.com/open62541/open62541/wiki/Capturing-traffic-with-Wireshark

We now can simulate deterministic timings during the unit tests. So when we get a traffic dump, we can reproduce the exact same conditions.

JinGuoGe, 2017-07-09:
Hi, @jpfr 
Sorry for delay. I 've made test program which occurred issue. 
This is test program. (https://www.dropbox.com/s/6gccyuelodykwyh/report_to_github.zip?dl=0)
I tested eclipse milo as opcua client. (https://github.com/eclipse/milo)
Same problem was occurred when using other opcua client program which based on eclipse milo. (https://github.com/comtel2000/opc-ua-client).
This is screenshot of my testing environment.
[image]
OPCUaServer is running in ubuntu, and OPCUaClient is running in windows by running java program.
I think now it is enough to generate issue in your side, because i provided test opcua server code and also opc ua client tool.
In screenshot, you can see i disabled network connection of ubuntu system.
Looking forward to your figuring problem reason.
Thanks.

jpfr, 2017-07-13:
Thanks. I could reproduce.
Fixed on my setup via #1111.

Please reopen if the issue remains.

closed 2017-07-13 by jpfr

JinGuoGe, 2017-07-15:
Hi, thanks.
I tested with updated version (#3702ffe7f3), but aborted by signal.
[image]
This is gdb log. 
https://paste2.org/Ua8cx4jF