open62541/open62541 #1092: Cannot connect to server with CTT

Infinity95, 2017-06-29:
Description
===========
With the current master branch i cannot connect to the server with the Compliance Test Tool anymore. Maybe this has something to do with the added ipv6 support? (Connecting with UAExpert works)

I'm using the config from the ctt repo: https://github.com/open62541/open62541-ctt
As far as i checked i don't think the server ever recieves any message.

Background Information / Reproduction Steps
===========================================
build master, run server and try running the ctt.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): fadd1dd09d8689e609339467a9c36651e19daa69
 - [ ] Other OPC UA SDKs used (client or server):  UA 1.02 Compliance Test Tool
 - [ ] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-06-29:
I cannot confirm. Works fine on my box with the current config from the repository.
Can you remove your CMake build directory and configure+build from scratch?
Sometimes it gets confused when files are renamed, and so on.

Infinity95, 2017-06-29:
I will check tomorrow, and see if that does the trick. (I thought i already did a clean build, but i'm not sure)
Although i find it strange, that this would only affect the CTT tool, since UAExpert works just fine.

Infinity95, 2017-06-30:
I just deleted my build directory and rebuilt from scratch. I then did a fresh clone of the open62541-ctt repo and loaded the config into the CTT. I then started the server and tried connecting with the CTT, and again all i got was a BadCommunicationError. 

Connecting with UAExpert still works. I also found out, that the client example cannot connect to the server as well. This results in the error message:
`[06/30/2017 13:13:42.143] warning/network       Connection to opc.tcp://localhost:16664 failed. Error: 0: Der Vorgang wurde erfolgreich beendet.`

Infinity95, 2017-06-30:
Ok, i found the reason it wasn't working. This was entirely my fail. I still had another opc tool running that was blocking the 4840 port, and since it was on autostart, it of course also blocked the port after restarting. Now that it's gone, everything works fine.

Infinity95, 2017-06-30:
Well, i only seem to have fixed half the Problem. I can now connect with the server and everything works fine, when i use 8c19500510e510094196b534064c7f5fb8c7d874. As soon as i try using the newest master fadd1dd09d8689e609339467a9c36651e19daa69, i still cannot connect with the CTT or the example client.

Another strange thing i encountered was, that when i try to build fadd1dd09d8689e609339467a9c36651e19daa69 with examples enabled, it sometimes fails the build and only works after buildng a second time. Also, when I then run the server, it sometimes fails with a memory access exception.
I always delete my build folder when switching branches, and i even went so far as to delete the complete repository and re-clone it.
I will check if this happens on a different machine as well, later or tomorrow.

jpfr, 2017-07-02:
Found it. The fix comes in two parts.

On windows, the server needs to listen on multiple sockets to support both IPv4 and IPv6.
This was added in #1094.
@janitza-thbe This goes in addition to #1084.
I added an explanation for IPv6 support to the CHANGELOG.

The 'localhost' target is resolved to an IPv6 address first.
But, the OPC Foundation ANSI C Stack did not fully support IPv6 until recently.
So the current CTT are not IPv6 ready.
https://github.com/OPCFoundation/UA-AnsiC/issues/10

Once the PR has landed, you can connect to the CTT on Windows with 127.0.0.1 (and not 'localhost').

jpfr, 2017-07-03:
#1094 has landed.

closed 2017-07-03 by jpfr