open62541/open62541 #1093: error in visual stuidio after enabling GENERATE NAMESPACE0

abuzer619, 2017-07-02:
Hello Everyone. 
I am using open62541 v0.2. As soon as i enable GENERATE NAMESPACE0 in CMakeList.txt after generating the build files and building the project in the visual studio, i get warnings such as :Can't ignore node, Namespace does currently not contain a node with id i=2292	and errors such as 'cmd.exe' exited with code 1. Can anyone please help?

jpfr, 2017-07-02:
@ichrispa Your call.

abuzer619, 2017-07-03:
any additional changes to be made in the Cmakelist.txt other than what showed in tutorial?

ichrispa, 2017-07-03:
Hi @abuzer619,

could you post (attach) the complete build log so i can have a look?

To clarify: the build fails because of cmd.exe returning an error code? Or does it fail later on?

Did you successfully build with an older stack version? If so, can you give me the approximate time or even better the commit id?
Not finding 2292 means the server object node (I think) is not in the XML, which is OK since our stack creates it.

Regards,

Ichrispa

abuzer619, 2017-07-04:
@ichrispa  Thank you for your reply. build fails because of cmd error. it didnt work on older version too. This is the log
1>------ Build started: Project: ZERO_CHECK, Configuration: Debug Win32 ------
1>  Checking Build System
1>  CMake does not need to re-run because C:/Users/MuhammadAbuzer/Desktop/binnew/CMakeFiles/generate.stamp is up-to-date.
1>  CMake does not need to re-run because C:/Users/MuhammadAbuzer/Desktop/binnew/doc/CMakeFiles/generate.stamp is up-to-date.
2>------ Build started: Project: open62541-object, Configuration: Debug Win32 ------
3>------ Build started: Project: open62541-plugins, Configuration: Debug Win32 ------
4>------ Build started: Project: open625451_amalgamation, Configuration: Debug Win32 ------
2>  Building Custom Rule C:/Users/MuhammadAbuzer/Desktop/open62541-0.2/CMakeLists.txt
2>  CMake does not need to re-run because C:/Users/MuhammadAbuzer/Desktop/binnew/CMakeFiles/generate.stamp is up-to-date.
2>  Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h, src_generated/ua_types_generated_handling.h, src_generated/ua_types_generated_encoding_binary.h
2>  Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h, src_generated/ua_transport_generated_handling.h, src_generated/ua_transport_generated_encoding_binary.h
3>  Building Custom Rule C:/Users/MuhammadAbuzer/Desktop/open62541-0.2/CMakeLists.txt
3>  CMake does not need to re-run because C:/Users/MuhammadAbuzer/Desktop/binnew/CMakeFiles/generate.stamp is up-to-date.
3>  Generating src_generated/ua_nodeids.h
4>  Building Custom Rule C:/Users/MuhammadAbuzer/Desktop/open62541-0.2/CMakeLists.txt
4>  CMake does not need to re-run because C:/Users/MuhammadAbuzer/Desktop/binnew/CMakeFiles/generate.stamp is up-to-date.
4>  Generating src_generated/ua_nodeids.h
2>  Generating src_generated/ua_namespaceinit_generated.c, src_generated/ua_namespaceinit_generated.h
3>  Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h, src_generated/ua_types_generated_handling.h, src_generated/ua_types_generated_encoding_binary.h
4>  Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h, src_generated/ua_types_generated_handling.h, src_generated/ua_types_generated_encoding_binary.h
2>  INFO:__main__:Preprocessing C:/Users/MuhammadAbuzer/Desktop/open62541-0.2/tools/schema/namespace0/Opc.Ua.NodeSet2.xml
4>  Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h, src_generated/ua_transport_generated_handling.h, src_generated/ua_transport_generated_encoding_binary.h
4>  Generating src_generated/ua_namespaceinit_generated.c, src_generated/ua_namespaceinit_generated.h
3>  ua_network_tcp.c
4>  INFO:__main__:Preprocessing C:/Users/MuhammadAbuzer/Desktop/open62541-0.2/tools/schema/namespace0/Opc.Ua.NodeSet2.xml
2>  INFO:__main__:Parsing C:\Users\MUHAMM~1\AppData\Local\Temp\Opc.Ua.NodeSet2.xml_preProcessed-x_aahfvk.xml
4>  INFO:__main__:Parsing C:\Users\MUHAMM~1\AppData\Local\Temp\Opc.Ua.NodeSet2.xml_preProcessed-0a27dndm.xml
2>  INFO:__main__:Linking namespace nodes and references
4>  INFO:__main__:Linking namespace nodes and references
3>  ua_clock.c
3>  ua_log_stdout.c
3>  ua_config_standard.c
3>  Generating Code...
2>  INFO:__main__:Building datatype encoding rules
4>  INFO:__main__:Building datatype encoding rules
2>  INFO:__main__:Allocating variables
2>CUSTOMBUILD : warning : __main__:Can't ignore node, Namespace does currently not contain a node with id i=2292
2>CUSTOMBUILD : warning : __main__:Can't ignore node, Namespace does currently not contain a node with id i=274
2>  INFO:__main__:Generating Header
4>  INFO:__main__:Allocating variables
4>CUSTOMBUILD : warning : __main__:Can't ignore node, Namespace does currently not contain a node with id i=2292
4>CUSTOMBUILD : warning : __main__:Can't ignore node, Namespace does currently not contain a node with id i=274
4>  INFO:__main__:Generating Header
2>  Traceback (most recent call last):
2>    File "C:/Users/MuhammadAbuzer/Desktop/open62541-0.2/tools/pyUANamespace/generate_open62541CCode.py", line 174, in <module>
2>      generatedCode = ns.printOpen62541Header(ignoreNodes, args.suppressedAttributes, outfilename=basename(args.outputFile))
2>    File "C:\Users\MuhammadAbuzer\Desktop\open62541-0.2\tools\pyUANamespace\ua_namespace.py", line 713, in printOpen62541Header
2>      code = code + r.printOpen62541CCode(unPrintedNodes, unPrintedRefs);
2>    File "C:\Users\MuhammadAbuzer\Desktop\open62541-0.2\tools\pyUANamespace\ua_node_types.py", line 697, in printOpen62541CCode
2>      code = code + codegen.getCreateNodeBootstrap(self)
2>    File "C:\Users\MuhammadAbuzer\Desktop\open62541-0.2\tools\pyUANamespace\open62541_MacroHelper.py", line 294, in getCreateNodeBootstrap
2>      code.append(node.getCodePrintableID() + "->displayName = UA_LOCALIZEDTEXT_ALLOC(\"en_US\", \"" +  node.displayName() + "\");")
2>  TypeError: must be str, not bytes
2>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code 1.
4>  Traceback (most recent call last):
4>    File "C:/Users/MuhammadAbuzer/Desktop/open62541-0.2/tools/pyUANamespace/generate_open62541CCode.py", line 174, in <module>
4>      generatedCode = ns.printOpen62541Header(ignoreNodes, args.suppressedAttributes, outfilename=basename(args.outputFile))
[... cut ...]
Warning		__main__:Can't ignore node, Namespace does currently not contain a node with id i=274	open625451_amalgamation	C:\Users\MuhammadAbuzer\Desktop\binnew\CUSTOMBUILD	1	
Error	LNK1181	cannot open input file 'C:\Users\MuhammadAbuzer\Desktop\binnew\open62541-object.dir\Debug\ua_types.obj'	open62541	C:\Users\MuhammadAbuzer\Desktop\binnew\LINK	1	
Error	MSB6006	"cmd.exe" exited with code 1.	open62541-object	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets	171	
Error	MSB6006	"cmd.exe" exited with code 1.	open625451_amalgamation	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets	171

abuzer619, 2017-07-05:
when i turn Build Example Nodeset Compiler ON on cmake i get the following error 
CMake Error at examples/CMakeLists.txt:71 (add_executable):
  Cannot find source file:

    C:/Users/MuhammadAbuzer/Desktop/binnew/src_generated/nodeset.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error: CMake can not determine linker language for target: server_nodeset

Is it important to turn it on?

jpfr, 2017-07-06:
The fix is coming in #1100.

jpfr, 2017-07-06:
Works on my VS2017 now.

closed 2017-07-06 by jpfr

ichrispa, 2017-07-08:
Cool, thank you @jpfr.