open62541/open62541 #1102: UA_Client_readValueAttribute() returns unexpected error (0x80010000)

vdeters, 2017-07-06:
Description
===========
This happens for me when the Value attribute has value = Null.

Background Information / Reproduction Steps
===========================================
Call UA_Variant_new() and then read the value attribute of any node that has value = Null. UaExpert shows Value = Null, StatusCode = Good, DataType = Double, ValueRank = -1.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):0.2-rc2
 - [ ] Other OPC UA SDKs used (client or server): server: OPC UA ANSI C Demo Server v1.7.0 (Windows), other client: UaExpert 1.4.4-275
 - [ ] Operating system:RHEL 7.1
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-07-06:
The code segment for this is very short:
https://github.com/open62541/open62541/blob/0.2/src/client/ua_client_highlevel.c#L377

What do you propose should be returned instead?

We could treat reading the `value` attribute differently from the other attributes and return just an empty variant in that case.
Currently, all attribute types are treated the same. When reading for example a nodeid, there is no sensible default value when the server does not return a value.

vdeters, 2017-07-06:
> ua_client_highlevel.c#L377

I see. I just need to able to distinguish "no value" from some other error,
because "no value" as a valid condition for me. I see that
BAD_UNEXPECTEDERROR is returned for other conditions as well so I can't
assume that error implies "no value". Is there any way I can check for "no
value" without duplicating your __UA_Client_readAttribute() code?
Vince

jpfr, 2017-07-06:
Just use `UA_Client_Service_read`.
It returns a `UA_ReadResponse` with an array of `UA_DataValue` inside.
So you have access to the full response from the server. Including statuscodes, timestamps, etc.

Does that work for you?

vdeters, 2017-07-06:
Actually I was reading the value attribute just to get its type. I've
changed that to get the type using UA_Client_readDataTypeAttribute() and
UA_findDataType() instead, which is working well enough. I do have one case
where UA_findDataType() returns null - what can I infer from that?

closed 2017-07-08 by vdeters