open62541/open62541 #1103: NodeSet compiler produces types that are not defined in open62541.h

XB32Z, 2017-07-06:
Hi, 

On branch master, the amalgamated open62541.c/.h make a difference between internal and external headers. The problem is that when using the code generator script, some types (like UA_MethodNode) are part of the header_internal hence are only amalgamated in open62541.c

This result that the generated files cannot be compiled. I solved this by removing the internal_header cmake variables from the open62541.c amalgamation call to the one from open62541.h but I guess there's some work to do to define which header can be moved there. Otherwise we might need to consider having a special install set for using with the generator.

Best,

jpfr, 2017-07-06:
Hello @XB32Z ,

Can you clarify the issue a bit more?

Which version of open62541 are you using?
What is the generator you refer to? The XML nodeset compiler?

The UA_MethodNode structure is hidden for a reason.
It's an implementation detail that is not exposed by the public API.

XB32Z, 2017-07-10:
Hey, 

The [issue](https://github.com/open62541/open62541/issues/1106) describes quite well the problem. 

I agree that the UA_MethodNode shouldn't be part of the public API, however, the file generated by the nodesetcompiler (from XML) are using this, hence we need to have it somewhere even on the install. 
Best,

XB32Z, 2017-07-10:
Ok I was looking a bit in this. As I see this, this could be transform into a new feature. When we model as XML then use the open62541, we can access the data outisde nodeset.c because in nodeset.h there are defines with the node ID of the data. However, for methods, there's no way to link a method to any implementation without modifying nodeset.c directly. 

What would be nice maybe is to replace the UA_MethodNode used currently by a UA_Server_addMethodNode (from the public API). Maybe we could imagine another method than nodeset(UA_Server*) which takes a map of nodeID and data access/set or methodCallback so that when the nodeset is compiled, the callback are directly set. 

What do you think?

XB32Z, 2017-07-24:
Hi, 
I saw that you updated this by changing in the python compiler from including the amalgamated open62541.h file to including some internal header. However when I use an installed version of open62541 those files are not installed. Hence I am still stuck here

Pro, 2017-08-13:
Related to #1138 

@santoshpanda1989 @XB32Z can you post your custom .xml file and a MVCE (https://stackoverflow.com/help/mcve) so we can reproduce the issue and try to solve it? Thanks

santoshpanda1989, 2017-08-15:
[Files.zip](https://github.com/open62541/open62541/files/1226086/Files.zip)
PFA the XML file which I am using to generate the node sets. I use QT to compile all the generated c files. I also have attached the QT project file and the generated c files.

XB32Z, 2017-08-23:
Hi all, 
Here's [mine](https://github.com/open62541/open62541/files/1245712/mytests.zip), if you run:

- mkdir build
- cd build
- cmake -Dopen62541_DIR:PATH=/some/path/to/install/lib/cmake ..
- make 

Or with the cmake GUI or similar on Windows.. 

Then the build will fail with error: servers/ua_server_internal.h no such file or directory. Because that file is not being copied in the install structure. Now we could either amalgamate it to open62541.h or have it separated and come up with some specific build for application using xml files as base (which to my experience is quite often)

Best, 
Pierre

Pro, 2017-09-07:
This will be fixed with #739.
I would be nice if you can test the version from #739 and leave any feedback there, if there are still open issues with your models.

An example on how to use the new nodeset compiler is shown in the examples dir:
https://github.com/open62541/open62541/blob/5cebb35d396513275ea5d6949ab6aba88c025010/examples/CMakeLists.txt#L74

Note that you may need to compile open62541 with the full NS0 (by default it is using a reduced namespace). For that use the CMake flags:
`cmake -DUA_NAMESPACE0_XML=../tools/schema/namespace0/Opc.Ua.NodeSet2.xml -DUA_DATATYPES_FILE= -DCMAKE_BUILD_TYPE=Debug -DUA_BUILD_EXAMPLES=ON ...
`
The empty `DATATYPES_FILE` tells the ns compiler to include all nodes.

Please leave any feedback directly within the #739 PR.

closed 2017-09-07 by Pro