open62541/open62541 #1106: Not able to compile XML nodeset

santoshpanda1989, 2017-07-07:
HI,
I am trying to compile my own XML file. I tried with cmake and then tried to compile the nodeset.c, open62541.c, and myServer.c but I am getting error while compiling it. Please find below the command.
gcc -std=c99 myServer.c nodeset.c open62541.c -o myServer

I have attached the screenshot of the error.

Thanks,
Santosh
[image]

jpfr, 2017-07-08:
@ichrispa That's a nodeset compiler issue.
Can you have a look?

santoshpanda1989, 2017-07-10:
Hi,
Is there any update?

Thanks,
Santosh

XB32Z, 2017-07-10:
Hi, 
I had this issue, it's the cmake that wasn't updated since v0.2 and that doesn't work with the master. I made a (issue already)[https://github.com/open62541/open62541/issues/1103]

Pro, 2017-08-13:
duplicate of #1103

closed 2017-08-13 by Pro

merlingeorge555, 2018-05-20:
Hi @santoshpanda1989 Can you teach me how to work on XML node set compiler?

merlingeorge555, 2018-05-20:
@santoshpanda1989  Thank you for your immediate response. Let me say my requirement.  
I have a XML file in standard NS0. 
Via cmake I should a create a .h and .c file and it should it produce a example server file. Is that possible? If you find the exact steps, Kindly share with me