open62541/open62541 #1107: Prosys client crashes with open62541 server (on master)

XB32Z, 2017-07-10:
Hi all, 

I am using the Java Prosys stack client to connect to my open62541 server. This worked for version 0.2 however on the master the client cannot find the endpoints and crashes with a EOF exception. 

I tried on different platforms, without certificates but it didn't solved the issue. Was there something changed on the lowest level since v0.2? 

Best,

jpfr, 2017-07-12:
Hey Pierre,

We are testing against the official Conformance Testing Tools of the OPC Foundation as well.
The tests are green. This is a screenshot of the tests for the "GetEndpoints" service on the current master branch.

[image]

A possible cause for this bug is the recent switch to dual IPv4 / IPv6. We now use IPv6 if available.

By the way, I don't understand the attitude of not fixing a crashing bug from a normal network packet.
Whether this is our "fault" or not.
The Prosys SDK should expect and protect against malicious input as industrial applications are getting targeted by hackers. Crashing bugs are at least a denial of service and could even be exploitable to take over the computer.

I have only a few minutes time today. Will investigate with the Prosys SDK itself at the end of the week.

XB32Z, 2017-07-12:
Hi, 

Thank you for your answer, I will investigate the IPv6 lead. I agree with you regarding the Prosys position. I hope they'll investigate it too. 

Best,
Pierre

XB32Z, 2017-07-19:
Hi, 
Did you have any success with this?
Thx

jouniaro, 2017-07-20:
I just updated the issue in Prosys OPC Forum, but I decided to copy it here as well, if you find it easier:

Might be an issue with Java, but with the current knowledge there is nothing else we can do. The socket is giving an EOFException, which is being raised properly as a ServiceResultException: Bad_CommunicationError. I cannot figure out any better error handling for this situation.

IMO, it is a bit exaggerated to label this as ‘crash’. The client application just fails to connect. OTOH, I am not sure what happens in the server.

Anyway, we are glad to help in sorting this issue out, if you find out more information about the problem. With the current knowledge we don’t know how to target this.

The Java stack should also support IPV6 in general, although it has not been used very extensively. I would find it better to support IPV4 as long as it is available, anyway.

Pro, 2017-07-20:
Crosslink:
https://forum.prosysopc.com/forum/opc-ua-java-sdk/cant-connect-to-open62541

jpfr, 2017-07-22:
Oh, the OPC UA maestro @jouniaro is here.

So the java.lang.NullPointerException was thrown by user-code and not inside the SDK.
I take back that point. For Java non-experts it takes some time to interpret those trace logs.

The issue was fixed via 414032bcf917557444166cb907c8dd38e1d04602.

We had a regression on `master` when the initial HEL message is split into several packets.
(Even though chunking is not allowed for HEL, the network may split packets however it wants.)
So the connection was closed even though we should have waited for more packets.
Actually, with the Prosys Client, the HEL message comes in 33 packets that need to be reasembled.

Here's a Wireshark screenshot of this behavior
[image]

The Prosys client works great now!

closed 2017-07-22 by jpfr

jouniaro, 2017-07-24:
Great to hear you managed to fix it and that it works both with IPV4 and IPV6.

The splitting of the packets is strange. AFAIK, the Java stack does not do anything special, but relies on standard Java libraries to perform the communication. So I have no idea why it splits that much in your environment.

(And yes, it's not always ((usually)) very easy to interpret the logs)