open62541/open62541 #1109: single-file distribution - Opc.Ua.NodeSet2.xml

aquari89, 2017-07-12:
Description
===========
Hello, All.
I am trying to configure an OPC UA Server using a single-file distribution(open62541.c, open62541.h).
There is one question.
How do I create a modeled node in Opc.Ua.NodeSet2.xml?
It is generated by using cmake in full source code.
But I do not know when to use single-file distribution.

Checklist
=========
Please provide the following information:

 - [ release 0.2 ] open62541 Version (release number or git tag):
 - [ Server ] Other OPC UA SDKs used (client or server): 
 - [ Ubuntu Linux ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

FlorianPalm, 2017-07-12:
Hey, 
you have to use the nodeset compiler to generate c-code from your XML nodeset. Afterwards you can include it in your application.
Have a look at ``server_nodeset.c`` , ``server_nodeset.xml`` and the ``CMakeLists.txt`` within the example folder (line 70 and following, invocation of the nodeset compiler)
I am not complete sure if I understood what you are trying to do.

Pro, 2017-08-13:
@aquari89 I assume this is resolved.
If not please let us know how we can help you further!

closed 2017-08-13 by Pro