open62541/open62541 #1112: CTT on master fails

Infinity95, 2017-07-14:
Description
===========
The CTT fails on test 10, Secuirty User Name Password:
[image]

The expected result is Bad_IdentityTokenInvalid, since the encryption algorithm is expected to be null, but a garbage algorithmUri is supplied.

I think this plugin will need some reworking, as soon as we have the first security policy.

Background Information / Reproduction Steps
===========================================
Checkout 20440bcd6b73a92270e3a4a85b493641558c0037 and run the ctt.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 20440bcd6b73a92270e3a4a85b493641558c0037
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2017-07-14:
#1113 is a quick fix of this. This will only work for a user token policy that does not encrypt (as it is currently).

jpfr, 2017-07-19:
Merged. Will there be a "complete" solution once we have encryption?

Infinity95, 2017-07-20:
Once we have the first security policy for encryption (not only the plugin api), the accesscontrol can and should be expanded to allow encryption. Then we also have to extend those checks to conform with the standard.

Pro, 2018-02-01:
Related to #934 (or duplicate?)

jpfr, 2019-02-10:
The client supports this on master.
Server-side support is coming in #2390.

PR #1113 (2017-07-19): Fix/accescontrol default
Fixes #1112
commit b15ed3ae (2017-07-19): Fix/accescontrol default (#1113)

* Fixed mismatching function declaration and the resulting build error

* Fix #1112
commit b4f1c9e3 (2017-07-19): Fix/accescontrol default (#1113)

* Fixed mismatching function declaration and the resulting build error

* Fix #1112
closed 2019-11-21 by jpfr