open62541/open62541 #1115: Integrate Fuzzing?

Pro, 2017-07-16:
Would it make sense to integrate the Google Fuzzing project?
https://github.com/google/oss-fuzz

It could be helpful to find security issues...

The main question is, how much work it would be to create some basic fuzzing targets and how well we can split the targets to cover open62541's code.

Any ideas from your side?

jpfr, 2017-07-16:
A proper integration of oss-fuzz would be very cool!

I ran afl from time to time on the decoding code.
http://lcamtuf.coredump.cx/afl/

For this, there is some infrastructure to parse binary messages from files.
https://github.com/open62541/open62541/blob/0.2/tests/check_server_binary_messages.c
https://github.com/open62541/open62541/tree/0.2/tests/dumps

I don't know how much effort the integration into oss-fuzz is.
What would speed up testing is some setup where we don't have to restart the server executable for every test.

Pro, 2017-07-18:
See:
https://github.com/google/oss-fuzz/pull/721

and https://github.com/open62541/open62541/tree/feature/fuzzing for some first tests.

I added the E-Mail address of the core developers to the fuzzer config. Let me know if I should remove someone:
https://github.com/Pro/oss-fuzz/blob/2f8b75ec515dbad4ab74b9db9ef2bd72d6e4a653/projects/open62541/project.yaml

Pro, 2017-07-20:
Copied from PR #1120:

What are useful targets for fuzzing?

    Decoding in general (don't need to start a server for that)
    Message Chunking (don't need to start a server either)
    Decoding the payload for specific services (needs the server when the service gets executed)

At a later point, we might add fuzzing of timings. For timeouts, and so on.
On master, the tests have full control over a simulated system clock.
But I'm not sure how this can be integrated into the fuzzing-framework.

Pro, 2017-08-13:
Basic fuzzing is integrated with #1120 and #1143 and #1147

closed 2017-08-13 by Pro