open62541/open62541 #1118: After timeout of session, get aborted

JinGuoGe, 2017-07-18:
Description
===========
Hi, @jpfr 
Thanks for fixing issue #1071 .
I tested with updated version 3702ffe. Crashing issue was solved but instead of it, get aborted.
This is gdb log.
https://paste2.org/Ua8cx4jF
This is my testing environment screenshot.
https://user-images.githubusercontent.com/15504060/28234883-6662c24a-6937-11e7-9fcf-062735af565e.png

Background Information / Reproduction Steps
===========================================
About reproduction steps, it is exactly same with reproduction test programs which i provided in #1071 

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 3702ffe
 - [ ] Other OPC UA SDKs used (client or server):  eclipse milo
 - [ ] Operating system: ubuntu 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-07-18:
Again, how to reproduce?
This happens with the same client and pulling out the network cable as for #1071?

JinGuoGe, 2017-07-18:
Yes, but reproduction process is exactly same. But issue is different. I think crashing issue was solved. Last issue was crash issue, but this issue is abort issue, not crashed.

JinGuoGe, 2017-08-04:
@jpfr 
How are you? Could you tell me reason and way to solve this issue?
Thanks.

StalderT, 2017-11-03:
duplicate of https://github.com/open62541/open62541/issues/1284 ?

StalderT, 2018-01-15:
@JinGuoGe Is the problem still present with the last master commit?

Pro, 2018-02-01:
Closing as it is probably fixed by latest master commits from @StalderT

StalderT, 2018-02-02:
@Pro No my commits doesn't fix this issue. I found a way to reproduce the issue. Can you reopen ?

StalderT, 2018-02-02:
@jpfr I need your help : 

Step to reproduce : 
1) Modifiy ```src/server/ua_subscription.c``` : 
```
static int test=0;
void
UA_Subscription_publishCallback(UA_Server *server, UA_Subscription *sub) {
    UA_LOG_DEBUG_SESSION(server->config.logger, sub->session,
                         "Subscription %u | Publish Callback",
                         sub->subscriptionId);
test++;
if (test>5){
                UA_Session_deleteSubscription(server, sub->session,
                                              sub->subscriptionId);
   return;
}
```

2) Start UaExpert and monitor "current date" node. After 5 publish the server crash (sometime) with : 

```
==13500== Invalid read of size 8
==13500==    at 0x159B28: UA_MonitoredItem_SampleCallback (ua_subscription_datachange.c:245)
==13500==    by 0x13DC36: UA_Server_workerCallback (ua_server_worker.c:125)
==13500==    by 0x151046: UA_Timer_process (ua_timer.c:300)
==13500==    by 0x13DEA0: UA_Server_run_iterate (ua_server_worker.c:332)
==13500==    by 0x13E072: UA_Server_run (ua_server_worker.c:428)
==13500==    by 0x11517F: main (server.c:458)
==13500==  Address 0x5c3acb0 is 16 bytes inside a block of size 168 free'd
==13500==    at 0x4C2CDDB: free (vg_replace_malloc.c:530)
==13500==    by 0x158D5A: MonitoredItem_delete (ua_subscription_datachange.c:52)
==13500==    by 0x15659F: UA_Subscription_deleteMembers (ua_subscription.c:48)
==13500==    by 0x154726: UA_Session_deleteSubscription (ua_session.c:113)
==13500==    by 0x157268: UA_Subscription_publishCallback (ua_subscription.c:290)
==13500==    by 0x13DC36: UA_Server_workerCallback (ua_server_worker.c:125)
==13500==    by 0x151046: UA_Timer_process (ua_timer.c:300)
==13500==    by 0x13DEA0: UA_Server_run_iterate (ua_server_worker.c:332)
==13500==    by 0x13E072: UA_Server_run (ua_server_worker.c:428)
==13500==    by 0x11517F: main (server.c:458)
==13500==  Block was alloc'd at
==13500==    at 0x4C2DBC5: calloc (vg_replace_malloc.c:711)
==13500==    by 0x158BDB: UA_MonitoredItem_new (ua_subscription_datachange.c:21)
==13500==    by 0x166836: Operation_CreateMonitoredItem (ua_services_subscription.c:262)
==13500==    by 0x13DB36: UA_Server_processServiceOperations (ua_server_utils.c:407)
==13500==    by 0x166BEE: Service_CreateMonitoredItems (ua_services_subscription.c:325)
==13500==    by 0x161B1B: processMSG (ua_server_binary.c:568)
==13500==    by 0x161D4D: processSecureChannelMessage (ua_server_binary.c:607)
==13500==    by 0x1539D7: UA_SecureChannel_finalizeChunk (ua_securechannel.c:731)
==13500==    by 0x1541C4: UA_SecureChannel_processChunk (ua_securechannel.c:997)
==13500==    by 0x162128: processCompleteChunk (ua_server_binary.c:728)
==13500==    by 0x151755: processChunk (ua_connection.c:143)
==13500==    by 0x1517F9: UA_Connection_processChunks (ua_connection.c:168)
```

The problem is that ```UA_Subscription_publishCallback``` is called here

https://github.com/open62541/open62541/blob/4c30ec02e9a90c86b11f3ae4dbbaf14fa97a91ec/src/ua_timer.c#L297

```UA_Subscription_publishCallback``` unregister sample callback by calling ```UA_Server_removeRepeatedCallback(server, mon->sampleCallbackId)``` : 

https://github.com/open62541/open62541/blob/4c30ec02e9a90c86b11f3ae4dbbaf14fa97a91ec/src/server/ua_subscription_datachange.c#L299

But the repeated job (sampling callback)  is still present in the list of callbacks to process now

https://github.com/open62541/open62541/blob/4c30ec02e9a90c86b11f3ae4dbbaf14fa97a91ec/src/ua_timer.c#L290

and cause the issue when it is called.

I do not know how to solve the problem. Calling ```processChanges(t);``` here : 

https://github.com/open62541/open62541/blob/4c30ec02e9a90c86b11f3ae4dbbaf14fa97a91ec/src/ua_timer.c#L298

do not solve the problem. Do you have an idea?

StalderT, 2018-02-02:
@jpfr https://github.com/open62541/open62541/pull/1570 fix the issue but I do not know if it's good in a multithreaded environment.

PR #1577 (2018-02-03): fix #1118
fix #1118

replace https://github.com/open62541/open62541/pull/1570
closed 2018-02-03 by jpfr