open62541/open62541 #1119: References are not present for Objects in the XML nodeset compiler process

santoshpanda1989, 2017-07-19:
Hi,
The namespace_init_generated.c file for an XML does not contain all the reference types for an object.
PFB the files and check for "firstscrewdriver". Please check and let me know.
Thanks,
Santosh
[Files.zip](https://github.com/open62541/open62541/files/1158887/Files.zip)

XB32Z, 2017-08-23:
Hi, 

I didn't try this, but I got a issue that is somewhat related. I have for some object the wrong TypeDefinition. I looked into the generated code and it seems that the namespace ID used is the wrong one.

Pro, 2017-08-23:
@XB32Z can you post some more detail, e.g. steps/code to reproduce your problem and debug info.
Otherwise it's quite hard to find the problem.

XB32Z, 2017-08-23:
Hi @Pro , 

Unfortunately I cannot share the files which are producing the errors.

However I made some other [examples](https://github.com/open62541/open62541/files/1245955/examples.zip) that you can try which provoke already quite some errors and end up to be partial or wrong in the server namespace. The design are quite complicated but I made them using UaModeler so it should be OPC UA correct.

I which I could help but the parser is quite big. 

Thx,

Pro, 2017-09-14:
This is fixed in the new nodeset compiler #739

closed 2017-09-14 by Pro