open62541/open62541 #1121: Implement VariableTypes from the DataAccess Spec (Part 8)

jpfr, 2017-07-20:
>     The DataAccess model extends the variable model by defining VariableTypes. The
>     DataItemType is the base type. ArrayItemType, AnalogItemType and DiscreteItemType (and its
>     TwoState and MultiState subtypes) are specializations.

The DataAccess Spec is built on top of the core OPC UA functionality.
We can provide the DataAccess VariableTypes in open62541.
With a build-flag to make them optional.

fmeerkoetter, 2017-08-11:
Can you please give a (short) outline what would need to be done?

fmeerkoetter, 2017-09-14:
@Pro Can you provide an outline for this?

Pro, 2017-09-14:
I'm not 100% sure what @jpfr means, maybe he can shed more light on it?

Probably adding the addtional datatypes which are mentioned in the spec. The datatypes in the NodeSet2.xml will be added anyways when using the new nodeset compiler in #739. So IMHO these "new" VariableTypes should already be there?

The corresponding spec can be found here:
https://opcfoundation.org/developer-tools/specifications-unified-architecture/part-8-data-access/

Stasik0, 2017-09-14:
I think codesys rte provide a plcopen view. They run for 2 hours in a trial mode.

Stasik0, 2017-09-14:
Wrong thread, sorry

Pro, 2018-02-01:
@jpfr is this still an open issue or fixed by creating the corresponding nodes from the full NS0?

jpfr, 2020-03-29:
This is done on master.

closed 2020-03-29 by jpfr