open62541/open62541 #1138: Unable to combine nodeset compiler and register server.

santoshpanda1989, 2017-08-04:
Hi,
I would like to combine the server_register and node set compiler features. But unfortunately, I am unable to do so.

Description
===========
I enabled the UA_ENABLE_DISCOVERY_MULTICAST and UA_ENABLE_GENERATE_NAMESPACE0 in the cmakeLists.txt and built it. The node sets are included in the open62541.c file. When I am building the Server with open62541.c and server_register.c, There isn't any error while building the open62541.c and server_register.c files together. However, when I connect to the server through a client UI(ProSys), I am not able to browse anything.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): Prosys Client
 - [ ] Operating system:OSX
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-08-06:
Note that `UA_ENABLE_GENERATE_NAMESPACE0` is currently in experimental stage.

Anyways, can you try to run the unit tests and see if those are running through?

```
cmake -DCMAKE_BUILD_TYPE=Debug -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_BUILD_UNIT_TESTS=ON -DUA_ENABLE_COVERAGE=ON -DUA_ENABLE_VALGRIND_UNIT_TESTS=ON -DUA_ENABLE_GENERATE_NAMESPACE0=ON ..
make -j && make test ARGS="-V"
```

If all the tests succeed, please try the following options and report back if any of those is working:

- Disable `UA_ENABLE_DISCOVERY_MULTICAST` and `UA_ENABLE_GENERATE_NAMESPACE0` and try to connect with ProSys
- Only enable `UA_ENABLE_DISCOVERY_MULTICAST` and try ProSys
- Only enable `UA_ENABLE_GENERATE_NAMESPACE0` and try ProSys

santoshpanda1989, 2017-08-06:
@Pro I tried by enabling only UA_ENABLE_DISCOVERY_MULTICAST, it is working fine. But unfortunately, UA_ENABLE_GENERATE_NAMESPACE0 is not working. I am using the server_nodeset.c file by changing the nodeset(server) to ua_namespaceinit_generated(server) in the file. But the UA_NAMESPACEINIT_GENERATED function is not recognized because the declaration is present in open62541.c file. I have tried the same with another release of open62541 and it is working ok. Also, I want to include UA_NAMESPACEINIT_GENERATED function in the server_register.c file.
I have attached the output while running the server.
[server_nodeset_output.txt](https://github.com/open62541/open62541/files/1203288/server_nodeset_output.txt)

Pro, 2017-08-07:
The `UA_ENABLE_GENERATE_NAMESPACE0` seems to be currently broken and that's why it is declared as an experimental feature. I tested current master and it does not create any nodes. Mainly because of the errors:
```
[08/07/2017 07:45:40.062] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Reference type is not hierarchical
[08/07/2017 07:45:40.062] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference is invalid
[08/07/2017 07:45:40.063] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
```

Also note:
You do not have to use `ua_namespaceinit_generated(server)` directly. The generated namespace will be automatically used when you use the compile switch in the server internally.

We have to investigate the problem further to see where it fails.

santoshpanda1989, 2017-08-07:
@Pro I tried UA_ENABLE_GENERATE_NAMESPACE0 in release 0.2. I tried the method which I have mentioned earlier i.e. I tried to compile ua_namespaceinit_generated.c,open62541.c, 
[image]
[Server_nodeset_Output.pdf](https://github.com/open62541/open62541/files/1204523/Server_nodeset_Output.pdf)

and server_nodeset.c files. Now there is an error "Reference type is not hierarchical" but I am able to browse the nodes in ProSys. I have attached the server output and a screenshot of ProSys.

Pro, 2017-09-07:
This will be fixed with #739

closed 2017-09-07 by Pro