open62541/open62541 #1139: Network-TCP-Plugin ignores configured Logger

janitza-thbe, 2017-08-07:
Description
===========
Contrary to ua_network_udp.c the ua_network_tcp.c plugin does not use to standard logger from externally provided configuration any more. Instead it uses hard-coded the UA_Log_Stdout procedure from ua_log_stdout.h/.c make it impossible to use own logger formats or even syslog backends.

Checklist
=========
Please provide the following information:
 - [X] open62541 Version (release number or git tag): master/812bab5

jpfr, 2017-08-09:
Hello @janitza-thbe 

duly noted.
We will then change the interface to "create" the networklayer with a logger pointer.

In the past, the server forwarded the logger to the networklayer during startup.
Now we aim to have a "const" configuration that the server does not write into...
And the networklayer set up and ready when the server starts.

m0eter, 2017-11-08:
Hello @jpfr, @janitza-thbe ,

Just wanted to note, that this affects the client networklayer as well. 

With the server this is especially problematic as I could not figure out a way to see if UA_Server_run actually could open the TCP socket. If it is blocked, the Serverstatus Node just says running. The only way I found to check if the server is running correctly would be to use a local client for a test connect, and this is far from ideal.

Pro, 2018-02-01:
fixed in #1566

Thanks @janitza-thbe!

closed 2018-02-01 by Pro