open62541/open62541 #1149: Example for Embedded System

Gfast2, 2017-08-10:
Hello open62541,

After firing up the [OPC UA Server examples](https://github.com/open62541/open62541/tree/0.2/examples)  on Beaglebone Black (Raspberry Pi like) and viewing them by [UaExpert](https://www.unified-automation.com/de/produkte/entwicklerwerkzeuge/uaexpert.html), I was very impressed by open62541 projected and far more enough motivated  to port this library to my [ESP32](http://esp32.net/) based project. 
Because of its toolchain (freeRTOS + LWIP) and my skill set, I got stucked on implementating open62541 in this enviroment. What I found is only [this issue](https://github.com/open62541/open62541/issues/103) 🏷 in Github, which not tell the story to the end. 🐇 
Because of the trend - The embedded stuff is getting much performanter then before. It make more sense to port good libraries like open62541 not only for Linux in its example, but also for typical ones like esp32. 😃 
Is there any plan?

Pro, 2017-08-10:
Hi @Gfast2,
nice to see that you are also currently working on porting open62541 to freeRTOS + LWIP.

Interestingly, we are currently also trying the same and managed change to code in a way that open62541 compiles, at least in a quick and dirty approach. I will see if we can push it to a repository so that you can also test it and improve it.

Additionally, there is also this PR: #1141

Gfast2, 2017-08-10:
@Pro ,
*cool*! Compare with MQTT, OPC UA ist still not adopted widely on big IoT Platform like [AWS](https://aws.amazon.com/iot/). I think one of the big problem is its licensing nature, but not its core idea/design. For the much more serious IIoT (Industrial IoT) applications, OPC UA has to be the way of the further.

---

_(Right now I was just drunk by figuring out how many RTOSes out there by [this post](https://www.quora.com/Is-anyone-still-using-a-proprietary-embedded-OS-like-VxWorks-QNX-ThreadX-etc-on-production-systems-or-has-the-world-moved-onto-embedding-Linux-onto-all-platforms))_

---

OK, I'll give my best to tracking the working progress of [#1141](https://github.com/open62541/open62541/pull/1141), hopefully I will do some dedication for it. If I not, I will still leave a eye on this. May be somehow later. 

_ps:_
[Zephyr](https://www.zephyrproject.org/about/what-is-zephyr-project) from [Linux Foundation](https://www.linuxfoundation.org/) looks like another _Linux_ for embedded system. And it should have the common sense like open62541 project - Commertially Neutral, Open-Source and even based on a bullet-prooft Wind River's commercial VxWorks Microkernel. May be I'll do some dedication for open62541 with this Plattform.

At last thanks for your reply. I think I'd better close this _"Issue"_ first. 👌

closed 2017-08-10 by Gfast2

Pro, 2017-08-16:
@Gfast2 here is the PR for FreeRTOS: https://github.com/open62541/open62541/pull/1160

It would be nice if you can test it on your hardware.
For now you need to change the hardcoded IP of the server here:
https://github.com/open62541/open62541/pull/1160/files#diff-75a03bcf91e9852cb1cf8401c2586760R405

Gfast2, 2017-08-16:
@Pro I'll tell you the result later.

Pro, 2018-02-23:
Note that support for ESP32 will be added through the PR #1595 (tested and working with my Adafruit ESP32 Board)

Pro, 2018-04-17:
My example project for OPC UA on an ESP32 using Arduino is now available here:
https://github.com/pro/open62541-arduino