open62541/open62541 #1173: Bad formatted LocaleId in the DisplayName attribute of nodes

jonasgreen88, 2017-08-22:
Description
===========
The LocaleId of the DisplayName attribute of nodes in the stack separates the language and country/region with an underscore. According to the OPC UA Part 3, Chapter 8.4 LocaleId, the country/region code shall be preceded by a hyphen.

Also, is it really necessary to add LocaleId to nodes representing e.g. types? For types the DisplayName attribute (always?) contains the type name and a name usually doesn't change depending on the language. The UA Sample Servers from the OPC Foundation (https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications) even omits the LocaleId of the DisplayName attribute for all nodes what I can see.

I can publish a PR if we can decide how to solve this issue.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ x ] open62541 Version (release number or git tag): 0.2
 - [ x ] Other OPC UA SDKs used (client or server):  UA Expert client
 - [ x ] Operating system: Windows 7 / embedded c
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: Not required
 - [ ] Wireshark network dump attached: Not required
 - [ ] Self-contained code example attached: Not required
 - [ ] Critical issue: No

Pro, 2017-08-22:
Thanks for pointing that out. Looks like we are using underscores in many places where there should be hyphens for the LocaleId. It would be nice if you can create a PR for that.

Regarding the DisplayName for types:
You probably mean this line here (which should also be fixed to use hyphen):
https://github.com/open62541/open62541/blob/0.2/src/server/ua_server.c#L360

I'm not sure if it is allowed to set the `LocaleId` to Null. Unfortunately that's not clarified in the spec.
It would definitely safe some bytes, but the question is if it is still valid. Probably yes?

jpfr, 2017-08-22:
👍 For not using locales where we do not address a specific language (only).

HolgerJeromin, 2017-08-22:
I really love [standards](https://www.xkcd.com/927/).  :)
ISO 15897/POSIX locales has _ but ISO 639/BCP 47 has -

Pro, 2017-08-22:
So let's create a new standard, and use bang `!` instead :1st_place_medal: 

Joke aside, I was also confused about that, but after reading the Spec once again they state to use ISO 639

[image]

jonasgreen88, 2017-08-23:
> I'm not sure if it is allowed to set the LocaleId to Null. Unfortunately that's not clarified in the spec. 
> It would definitely safe some bytes, but the question is if it is still valid. Probably yes?
In OPC UA Part 3, chapter 8.4 LocaleId, it says: "An empty or null string indicates that the LocaleId is unknown."
Chapter 8.5, LocalizedText, doesn't say anything that would override that I think.

I think you all agree to changing underscores to hyphens to conform to the ISO standard. But how about set the LocaleId to null or an empty string where a specific language is not addressed?

Pro, 2017-08-23:
We can safely remove the locale where we don't address a specific language (which should be almost everywhere where we currently use the locale).
I just checked the Demo Prosys Server `opc.tcp://uademo.prosysopc.com:53530/OPCUA/SimulationServer` and it also does not set the locale.

jpfr, 2017-09-09:
#1182 was merged. Can we close?

closed 2017-09-09 by Pro