open62541/open62541 #1175: Application description not changed for endpoint

Pro, 2017-08-22:
If you create a configuration for the server and change the application description, it is only changed for the server, but not for its endpoints.

E.g.:
```
UA_ServerConfig *config = UA_ServerConfig_new_default();

UA_String_deleteMembers(&config->applicationDescription.applicationUri);
UA_LocalizedText_deleteMembers(&config->applicationDescription.applicationName);

config->applicationDescription.applicationUri = UA_String_fromChars("My-app-uri");
config->applicationDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("en-US", "My opc ua app");

UA_Server *server = UA_Server_new(config);

UA_StatusCode retval = UA_Server_run(server, &running);
UA_Server_delete(server);
UA_ServerConfig_delete(config);
```

Before commit https://github.com/open62541/open62541/commit/a8012d3040f903ffd127d06c70c73cbc87379602 the app description was changed for both.

You can simply test this with UA Expert:
[image]

The cause for this is that within `UA_ServerConfig_new_default` the app description is copied to the endpoint, instead of a reference assignment:
https://github.com/open62541/open62541/commit/a8012d3040f903ffd127d06c70c73cbc87379602#diff-1460dbbb030b9af913afd8afa5e4092cR88

@jpfr @Infinity95 is this intended behaviour?

I would expect that if you change the main app description, it is also changed for all the endpoints.

Infinity95, 2017-08-22:
I believe it should be possible to set this per endpoint. Each endpoint has a applicationURI, that must match with the certificate for that endpoint, so i'd think it makes sense to have a separate description for each endpoint as well.

Pro, 2017-08-24:
Ok, thanks for the info! Then let's go for that option that you need to change the endpoint description too.

closed 2017-08-24 by Pro

Infinity95, 2017-08-24:
Maybe the application description should be removed from the config entirely, so that it only resides in the endpoint. That might avoid some confusion?

Pro, 2017-08-24:
The application description in the config is still needed I think.
At least UA Expert shows in the "Add Server" Window first the Application Name from the config, and then when you select an endpoint and add the server, it shows the application name from the endpoint.

jpfr, 2017-08-24:
The application description is copied to a node in the namespace as well.
https://github.com/open62541/open62541/blob/master/src/server/ua_server_ns0.c#L524

I guess we have to support duplicate descriptions as they are endpoint-specific with regards to certificates.