open62541/open62541 #1180: Verify encoding/decoding of enums

jpfr, 2017-08-25:
Enums in C have at least the size of ``int``.
This can be 16bit on some architectures.
Currently, we just copy a 32bit value for the wire protocol.

We can add more complications to the `UA_DataType` descriptions.
Or we use int32_t instead of enums.

The current proposal is change the enum definitions from
```c
typedef enum {
    UA_NODECLASS_UNSPECIFIED = 0,
    UA_NODECLASS_OBJECT = 1,
    UA_NODECLASS_VARIABLE = 2,
    UA_NODECLASS_METHOD = 4,
    UA_NODECLASS_OBJECTTYPE = 8,
    UA_NODECLASS_VARIABLETYPE = 16,
    UA_NODECLASS_REFERENCETYPE = 32,
    UA_NODECLASS_DATATYPE = 64,
    UA_NODECLASS_VIEW = 128
} UA_NodeClass;
```
to
```c
#define UA_NODECLASS_UNSPECIFIED 0
#define UA_NODECLASS_OBJECT 1
#define UA_NODECLASS_VARIABLE 2
#define UA_NODECLASS_METHOD 4
#define UA_NODECLASS_OBJECTTYPE 8
#define UA_NODECLASS_VARIABLETYPE 16
#define UA_NODECLASS_REFERENCETYPE 32
#define UA_NODECLASS_DATATYPE 64
#define UA_NODECLASS_VIEW 128
typedef UA_Int32 UA_NodeClass
```

This has no impact on the code besides the python-based datatype generation.
Comments and ideas @open62541/open62541-core (and everybody else)?

jpfr, 2017-08-25:
Also, this can be backported to 0.2....

Pro, 2017-08-25:
Looks good to me! The proposed solution is even one line shorter and does not increase complexity 👍

Stasik0, 2017-08-25:
we can also use static asserts to verify the witdh i.e. sizeof(UA_NodeClass) == sizeof(UA_Int32)

some black-magic here: https://stackoverflow.com/a/3385694/383834

jpfr, 2017-08-25:
Yeah for @Stasik0's proposal to force 32bit enums. 😄

Pro, 2017-09-15:
I created a PR which enforces 32bit enums using static assert.
See #1210

If you agree, we can backport it to 0.2

commit 68b7851f (2017-09-16): Fix #1180 ensure enums are 32bit integers

If that's not the case, the error during compile time will be similar to
`error: named bit-field 'static_assertion_failed_enum_must_be_32bit' has zero width`
commit b7ae3a13 (2017-09-16): Fix #1180 ensure enums are 32bit integers

If that's not the case, the error during compile time will be similar to
`error: named bit-field 'static_assertion_failed_enum_must_be_32bit' has zero width`
commit 68d93709 (2017-09-16): Fix #1180 ensure enums are 32bit integers

If that's not the case, the error during compile time will be similar to
`error: named bit-field 'static_assertion_failed_enum_must_be_32bit' has zero width`
commit c12c782f (2017-09-17): Merge pull request #1210 from open62541/hotfix/assert_enums

Fix #1180 ensure enums are 32bit integers
closed 2017-09-17 by Pro