open62541/open62541 #1181: User Identity token policies are not configurable

jonasgreen88, 2017-08-28:
Description
===========
The User identity token policies are defined in ua_server_internal.h (ANONYMPUS_POLICY and USERNAME_POLICY). They are used in ua_server.c when creating endpoint definitions in addEndpointDefinitions().

Is it intended to define them internally in the stack? Shouldn't it be possible to configure the policies per project that uses the stack? In that case these strings maybe should be part of the UA_AccessControl type which in turn is part of the UA_ServerConfig type?

Background Information / Reproduction Steps
===========================================
User identity token policies are not configurable without editing the source code of the stack.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-08-28:
Hello @jonasgreen88 ,

this is implemented on master.
The endpoints are being created by the user (or by a plugin implementation) in the server config.
The server itself does only check that at least one endpoint is available.
https://github.com/open62541/open62541/blob/master/src/server/ua_server.c

This is in preparation to encryption support.
Endpoints combine policies for security channels with authentication methods.

Does this answer the question?

jonasgreen88, 2017-08-28:
Hi,

Thanks, that looks good. Tried to look for it on the master branch but didn't really find it.

closed 2017-08-28 by jonasgreen88