open62541/open62541 #1190: UA_Client_readNodeClassAttribute returns 'An unexpected error occurred' (v0.2)

WurstSchnitte, 2017-09-04:
Description
===========
Hi,
when my client tries to get the class attribute of a node via UA_Client_readNodeClassAttribute the returned StatusCode is always: 'An unexpected error occurred' (numeric value = 2147549184).

I tried it on an empty(no own nodes added) open62541 Server and on a opc ua demo server (opc.tcp://opcua.demo-this.com:51210/UA/SampleServer)

Here is my simplyfied code:
```c++
#include <iostream>
#include "open62541.h"

std::string getNodeClass(const UA_NodeId& nodeId, UA_Client* client) {
    UA_NodeClass uaClass;
    UA_NodeClass_init(&uaClass);

    UA_StatusCode retval = UA_Client_readNodeClassAttribute(client, nodeId, &uaClass);
    if(retval != UA_STATUSCODE_GOOD) {
        std::cout << "Error at getNodeClass: " << UA_StatusCode_description(retval)->explanation << " (" << std::to_string(retval) << ")\n" << std::endl;
        return "";
    }

    switch(uaClass) {
        case UA_NODECLASS_UNSPECIFIED:
            return "Unspecified";
        case UA_NODECLASS_OBJECT:
            return "Object";
        case UA_NODECLASS_VARIABLE:
            return "Variable";
        case UA_NODECLASS_METHOD:
            return "Method";
        case UA_NODECLASS_OBJECTTYPE:
            return "ObjectType";
        case UA_NODECLASS_VARIABLETYPE:
            return "VariableType";
        case UA_NODECLASS_REFERENCETYPE:
            return "ReferenceType";
        case UA_NODECLASS_DATATYPE:
            return "DataType";
        case UA_NODECLASS_VIEW:
            return "View";
        default:
            std::cerr << "Unknown NodeClass" << std::endl;
            return "";
    }
}

int main() {
    std::string serverAddress = "opc.tcp://opcua.demo-this.com:51210/UA/SampleServer";

    UA_Client* client = UA_Client_new(UA_ClientConfig_standard);
    UA_StatusCode retval = UA_Client_connect(client, serverAddress.c_str());
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return -1;
    }

    std::cout << "\n" << "Object Folder: " << std::endl;
    UA_NodeId objectFolder = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    getNodeClass(objectFolder, client);

    std::cout << "\n" << "Server Object: " << std::endl;
    UA_NodeId serverObject = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER);
    getNodeClass(serverObject, client);

    std::cout << "\n" << "ServerStatus Variable: " << std::endl;
    UA_NodeId serverStatusVariable = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_SERVERSTATUS);
    getNodeClass(serverStatusVariable, client);

    std::cout << "\n" << "GetMonitoredItems Method: " << std::endl;
    UA_NodeId serverGetMonitoredItemsMethod = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_GETMONITOREDITEMS);
    getNodeClass(serverGetMonitoredItemsMethod, client);

    UA_Client_delete(client);
}
```

Here is the console output with UA_LOGLEVEL=100:
[output.txt](https://github.com/open62541/open62541/files/1274249/output.txt)

Background Information / Reproduction Steps
===========================================
I compiled the client with the Single-file distribution for Linux64 (open62541.c/.h)

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): v0.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux 64bit (Kubuntu)
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-09-05:
Thanks for the Bug Report. You can use a Decimal to Hex Converter to convert the numeric error code to a hexadecimal code. Then check the `include/constants.h` file and search for the hex code. This yields the error `UA_STATUSCODE_BADUNEXPECTEDERROR`.

I think that this was fixed with this commit, but only in `master`
https://github.com/open62541/open62541/commit/6ae9c485e126c62254f14f641900706ede072d45

@WurstSchnitte you can apply this commit manually until we backport it to `0.2`

@jpfr did you already fix this in your mentioned branch, or should we backport my commit?

jpfr, 2017-09-05:
@Pro Please backport what you have.

PR #1192 (2017-09-05): Fix reading of nodeClass attribute
Before it did return UA_BADINTERNALERROR.

Also fix #1190
closed 2017-09-05 by Pro