open62541/open62541 #1205: 0.3 release - open issues

Pro, 2017-09-14:
The 0.3 release is near.

There are currently the following open issues/pull requests which need to be merged.

The most important ones are:
- [ ] Encryption plugin structure (#1003, #1187)
- [ ] Basic encryption support
- [ ] Nodeset plugin (#1168)
- [ ] Nodeset compiler (#739)

And here the full list:
https://github.com/open62541/open62541/milestones/0.3

After these are integrated a 0.3 release candidate will be published.

@open62541/open62541-core any more points you would like to add? Feel free to change the milestone tag of the issues.

ccvca, 2017-09-15:
Should the async-Client (open62541:async_client_loop) be part of 0.3? If not, my issue (#1164) is not part of 0.3

Pro, 2017-09-15:
@ccvca thanks, I just removed it from the milestone

Pro, 2017-09-17:
See also #710

ctron, 2017-09-18:
As already mentioned on the 4diac mailing list ... I would be interested in bringing version 0.3 of open62541 into Fedora 28 (instead of currently 0.2).

The easiest deadline for this would be 2018-02-20, also see [1]. It should be possible to use a release candidate build of open62541 for this and do an update later on.

[1] https://fedoraproject.org/wiki/Releases/28/Schedule

jpfr, 2017-09-18:
Great to hear @ctron !
Given that the remaining features are nearly done in a PR, we could even aim for a v0.3.1 in Fedora 28.

Infinity95, 2017-11-12:
You could maybe add encryption support for the client to the list. Although the none policy currently works with the client, there will be some changes necessary in order for full encryption to work.
This currently has low priority on my list of todo's (first unit tests, a working Basic128Rsa15 policy and then a plugin structure for certificate validation). If you want me to prioritize anything other first, tell me 😃

sirop, 2018-01-23:
@Infinity95 
Is there a case of encrypted password sent to the server through non encrypted channel? As in https://github.com/open62541/open62541/issues/788#issuecomment-293891197 ?

Infinity95, 2018-01-23:
That isn not implemented yet.

Pro, 2018-02-01:
Closing as all the referenced features are in master.
0.3 release is near. As soon as all the open issues/PRs marked with 0.3 milestone are fixed we will release a Release Candidate.
We expect it to be within mid of February (and would appreciate any help on closing the remaining issues)

closed 2018-02-01 by Pro