open62541/open62541 #1211: 0.2.1 Maintenance release

jpfr, 2017-09-15:
The 0,2 branch is ~30 commits ahead of v0.2.
I would tag a maintenance release v0.2.1 in the coming days.

Do you have features on 0.3 that should be backported to a 0.2 maintenance release?
The following files can be "mined" for improvements:

- ua_types.c
- ua_types_encoding_binary.c

StalderT, 2017-09-15:
PR https://github.com/open62541/open62541/pull/1058

jpfr, 2017-09-18:
I'm a bit unsure about #1058.
It's a large commit that changes all services operating on the references.
We risk to introduce additional bugs by backporting this to 0.2.
And there is no bug per se. Just that adding many nodes gets slow after the first couple 10k.

Pro, 2018-11-06:
@open62541/open62541-core are we still planning a 0.2.1 release?

I would vote for focussing on 0.3 and neglect the changes on 0.2?

jpfr, 2018-11-18:
0.2.1 got tagged with the fixes that were already backported.

closed 2018-11-18 by jpfr