open62541/open62541 #1213: Python 3 Error

ccvca, 2017-09-16:
Description
===========
When I try to build open62541 with Python 3 (cmake chose the wrong version) I got the following error:
  Traceback (most recent call last):
    File "D:/ProgrammierenD/open62541_master0.3/tools/generate_datatypes.py", line 480, in <module>
      printh("#define " + outname.upper() + "_COUNT %s" % (str(len(filtered_types))))
  TypeError: object of type 'filter' has no len()

According to this https://github.com/raghakot/keras-vis/issues/24 python3 returns an iterator instead of a list(?).

Maybe force Python 2.7 by CMake and remove the hint "(Python 3.x should work, too)" from https://open62541.org/doc/current/building.html (quick fix)

Long term fix: Adapt Skript(s?) to work with python 2 and 3 (and test python3 builds by travis) 

Python Version 3.6.2rc2

Background Information / Reproduction Steps
===========================================
Target Platform Visual Studio 2017, when I select Python2.7 manually in CMake (advanced options enabled) everything compiles fine.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 5cba5d2161208acf57c7d2748f9f7b9764f81c2d
 - [ ] Other OPC UA SDKs used (client or server):
 - [x] Operating system: Win10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

HolgerJeromin, 2017-09-16:
Two years ago python 3 worked. But we have now more python code and no travis check against this version.

Pro, 2017-09-16:
thanks for reporting.
Acutally it was just one small bug in the `generate_datatypes.py`. the rest works on the fly.

This will be fixed with #1214 and travis will also check python3 in the future

commit f94a6280 (2017-09-16): Fix compilation with python3 and add python3 to CI

Fix #1213
closed 2017-09-16 by Pro