open62541/open62541 #1219: Client API - Race condition when using multiple clients in same program

clepelli, 2017-09-19:
Description
===========
I use 2 threads to connect the same OPC server : 
- first thread is used to make synchronous calls to read / write services
- second thread loops on manuallySendPublishRequest and reads the results.
Each of these has its own UA_Client so I thought it would be thread safe.

BUT 

When I perform many calls on both threads, one of them always ends up logging an error on the server response decoding (UA_STATUSCODE_BADDECODINGERROR).

Background Information / Reproduction Steps
===========================================

Workaround : 
I saw that some variables including pos and end are global but not defined thread local unless UA_ENABLE_MULTITHREADING is set. I think this is the root cause. When I force them to be thread local, I have no decoding errors anymore. I do not want to set UA_ENABLE_MULTITHREADING as it requires another dependency (urcu.h), so I modified this instead : 

`/* Thread Local Storage */
#ifdef UA_ENABLE_MULTITHREADING_LIGHT
# if defined(__STDC_VERSION__) && __STDC_VERSION__ >= 201112L
#  define UA_THREAD_LOCAL _Thread_local /* C11 */
# elif defined(__GNUC__)
#  define UA_THREAD_LOCAL __thread /* GNU extension */
# elif defined(_MSC_VER)
#  define UA_THREAD_LOCAL __declspec(thread) /* MSVC extension */
# else
#  warning The compiler does not allow thread-local variables. The library can be built, but will not be thread-safe.
# endif
#endif
#ifndef UA_THREAD_LOCAL
# define UA_THREAD_LOCAL
#endif`

I think these vars should always be marked as thread local, or at least there should be a warning in the doc (did I miss it ?) / at runtime about using multiple clients.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.2
 - [ ] Other OPC UA SDKs used (client or server):  /
 - [x] Operating system: windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-09-19:
Thanks for the kind words. 😄 

Do you use v0.2 (the last release) or the current HEAD of the 0,2 branch?
If the latter is the case, probably this commit is causing problems: 5b91f643c4410b5e6dc90c1cd0cccfa239659861

clepelli, 2017-09-19:
I believed I used the 0.2 release but you are right, I use something like current HEAD : I have actually downloaded the single file distribution from home page (https://github.com/open62541/open62541/releases/download/v0.2/open62541.c)

I thought this was the release... 

Anyway, I think you have identified the culprit commit.

jpfr, 2017-09-19:
The fix was merged.
You can download the last nightly build of the 0.2 branch as a drop-in upgrade.
https://open62541.org/releases/c9645d59b2.zip

Can you verify that the issues is resolved?
Otherwise we have to make a setup that reproduces the behaviour.

clepelli, 2017-09-20:
Hello, I have an issue with a previous commit :
\lib\open62541.h:3291:67:
 error: expected string literal before 'enum_must_be_32bit'
 UA_STATIC_ASSERT(sizeof(UA_BrowseResultMask) == sizeof(UA_Int32), enum_must_be_32bit);

I am not well aware of this topic, but it seems in the standard C++11 the enum_must_be_32bit should actually be a string litteral (with quotes) ?
I am using mingw-w64 on windows (targetting windows), with gcc 7.1.0

clepelli, 2017-09-20:
ok I actually use C++14 and the #ifdef checks on C version variable.

Pro, 2017-09-20:
This is probably related to https://github.com/open62541/open62541/pull/1216

@clepelli how exactly do you compile your source code? Can you post the command line and a very simple example code which reproduces the problem?

Pro, 2017-09-20:
My guess is that for the C++ call `_Static_assert` in line here:
https://github.com/open62541/open62541/pull/1216/files
is missing the quotes around msg

@jpfr we should add a unit test for C++ >=11 compilation

clepelli, 2017-09-20:
- mingw-w64 installation with gcc 7.1.0
- with constants for using c++14 and c11
- I call mingw32-make

This modif works with my config :
```
#if defined(__cplusplus) && __cplusplus >= 201103L
# define UA_STATIC_ASSERT(cond,msg) \
    static_assert(cond, #msg)
#elif defined(__STDC_VERSION__) && __STDC_VERSION__ >= 201112L /* C11 */
# define UA_STATIC_ASSERT(cond,msg) \
    _Static_assert(cond, #msg)
```
I tried other variants but only this one seems to work.

Pro, 2017-09-20:
@clepelli this is now fixed. See #1221 

Is your original issue still open or already fixed with #1220 ?
If fixed, please close this issue

clepelli, 2017-09-20:
Yes, the correction is effective. Thanks.

closed 2017-09-20 by clepelli