open62541/open62541 #1226: UA_Server_addDataSourceVariableNode() always returns UA_STATUSCODE_GOOD

jonasgreen88, 2017-09-25:
Hi,

I wonder why UA_Server_addDataSourceVariableNode() is implemented to always return UA_STATUSCODE_GOOD. Wouldn't it be good to get feedback if something fails when creating a data source variable node?

https://github.com/open62541/open62541/blob/2113ec1c9b4a96a17e1a8ee98952a71475911712/src/server/ua_services_nodemanagement.c#L966

I found that the function was changed to this behavior when fixing a memory leak in commit 72b0afbe7db885c381946e7782477aa05f81362d. But I'm not sure if the intention was to also change to always return success?

Pro, 2017-09-25:
Oh, yes, that is my fault. It should return `retval`

Pro, 2017-09-25:
Will be fixed with #1227

commit 957beede (2017-09-25): Merge pull request #1227 from open62541/hotfix/return_datasource_variable

Fix #1226 UA_Server_addDataSourceVariableNode always returned GOOD
commit ae16808b (2017-09-25): Fix #1226 UA_Server_addDataSourceVariableNode always returned GOOD
closed 2017-09-25 by Pro