open62541/open62541 #1230: reconnecting timeout

WoutvdBerg, 2017-09-27:
Description
===========
By removing cable between OPC-UA server and client the data won't be sent. The reconnecting function in the example runs client_connect but stays in this method for about 70 seconds before starting a new connection. 

Using the pause and resume button the connection will be restored immediately after the resume button has been pressed.

[connection timeout.pcapng.zip](https://github.com/open62541/open62541/files/1336315/connection.timeout.pcapng.zip)

Background Information / Reproduction Steps
===========================================
Windows 10 virtual machine with B&R simulator which runs the OPC-UA server, also tried with proses simulation server. Same problem.
Mac with open62541 client, example code: client loop connect.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3
 - [x] Other OPC UA SDKs used (client or server): Prosys simulation server, B&R simulator
 - [x] Operating system: Windows 10 and Os X
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-09-27:
Thanks for the report!
I tested the reconnect on Linux and Windows and it worked there. Maybe it's a Mac issue. I'll investigate that further.

In the meantime do you have the possibility to use open62541 client on a windows and/or linux machine to test the reconnect with the same server setup?

jpfr, 2017-09-27:
Hey all,

disrupting the connection by plugging out the cable is difficult to replicate.
Testing coverage is low as well.

Just killing the client process will gracefully close the network socket.
That is not the behaviour we want to test here.

My debugging-friendly setup is to use a virtual machine for the client (with a Linux or *BSD).
Then, the cable-plugging can be simulated by killing the driver with `service networking stop/start` or similar.
Any ideas for making this testable on Travis are very welcome.

What is the intended behaviour? Reconnect after the client timeout?
https://github.com/open62541/open62541/blob/0.2/plugins/ua_config_standard.c#L103

Pro, 2017-09-27:
My local tests looked like this:
The example mentioned above is just a loop which reads the server time from a remote OPC UA Server.

Stopping that server (using Ctrl+C) will trigger a connection error on the client side. As soon as the server is started again, the client will try to connect and read the time again. The reconnect timeout here is the same as in the config (https://github.com/open62541/open62541/blob/0.2/plugins/ua_config_standard.c#L103)
So this is a graceful shutdown of the server.

It is different if the network cable is unplugged. due to the nature of TCP it is quite hard to detect this case on client and server side, since it is on layer 2 and not layer 3.
See also here:

https://superuser.com/questions/911808/what-happens-to-tcp-connections-when-i-remove-the-ethernet-cable/911829

https://stackoverflow.com/questions/14782143/linux-socket-how-to-detect-disconnected-network-in-a-client-program

https://stackoverflow.com/questions/6404008/how-to-detect-a-tcp-socket-disconnection-with-c-berkeley-socket

So simple reconnect (restart of the server, which closes the TCP on shutdown) is already implemented in master branch, network disruptions are not yet handled.

Adding a test for this simple reconnect shouldn't be that hard (stop the server, restart it, and test if read is possible).

Testing a network cable unplug is not that easy and the question here is if we really want/need to test that or if it is just out of focus...

Pro, 2017-09-27:
I just remembered that I already implemented a unit test for reconnect:
https://github.com/open62541/open62541/blob/master/tests/check_client.c#L102

So if the server is shut down in a graceful way, reconnect should also work on OS X.
@WoutvdBerg can you please verify that? (I.e. just restart the OPC UA server in your VM while the loop example is up and running. It should be able to reconnect after latest 5 seconds).

jpfr, 2017-09-27:
We have a custom network layer for testing. The code is quite minimal as well.
https://github.com/open62541/open62541/blob/master/tests/testing_networklayers.c

If this becomes the source of more regressions in the future, we can add low-level networking unit tests.

WoutvdBerg, 2017-09-27:
I can confirm that the reconnecting is working but there is a ~72 seconds timeout when no action is taken. 

Stopping the server and starting the server reconnects within seconds, that's fine. 

Problem is when server and client are both running but the psychical connection is broken. My view on this is that there is a timeout built in somewhere (tcp protocol/windows/os x/open62541/OPC-UA) to re-establish the connection but it doesn't re-establish. After the timeout the reconnect is fine. It doesn't matter when the cable is plugged back in, after ~72-75 seconds the connection is back and gives the server time. I just want to shorten the timeout to seconds or make it possible to reconnect in the timeout.

WoutvdBerg, 2017-09-28:
I found the timeout, in os X it is called keepinit (net.inet.tcp.keepinit) standard value = 75000 (=75 seconds). Reducing this value results in earlier reconnecting.

The end of this timeout is start signal for the keep alive messages but I think because the connection is psychically broken the keepalive message results in ending the connection in the OS, after that open62541 can open a new connection.

Pro, 2017-09-28:
thanks for your further investigations and the solution.

The general problem with TCP is that it tries to send packages and caches them if it thinks that the connection is still up (so it didn't receive a RST), which is the case if the cable is just unplugged. The keepalive is used to detect this state and close the connection, which may take up to 2 hours.

It is also quite nicely described here:
http://coryklein.com/tcp/2015/11/25/custom-configuration-of-tcp-socket-keep-alive-timeouts.html

@open62541/open62541-core 
The general question for us is now:
Do we want to change the keepalive timeout and e.g. set it to the same value as the config by using these socket options, and at the same time overwrite the OS settings:

```
#include <sys/socket.h>
int keepalive_enabled = 1;
int keepalive_time = 180;
int keepalive_count = 3;
int keepalive_interval = 10;
setsockopt(socket_file_descriptor, SOL_SOCKET, SO_KEEPALIVE, 1, 
setsockopt(socket_file_descriptor, IPPROTO_TCP, TCP_KEEPIDLE, keepalive_time, sizeof keepalive_time);
setsockopt(socket_file_descriptor, IPPROTO_TCP, TCP_KEEPCNT, keepalive_count, sizeof keepalive_count);
setsockopt(socket_file_descriptor, IPPROTO_TCP, TCP_KEEPINTVL, keepalive_interval, sizeof keepalive_interval);
```

Or do we say that this setting is up to the user and he has to change it in the OS if he wants to make sure broken TCP connections are detected earlier? (This also means that we just close this issue with `wontfix`)

WoutvdBerg, 2017-09-28:
So the keep alive is used to maintain a connection which isn't used for a time, this starts after keepidle and then sends a empty packet, if no response this will be repeated till the keep alive_count.

My problem is with the keepinit, what does that timeout do except for delaying my reconnect. Is it usable to reconnect in some way? What are the risks of lowering the keepinit to for example 5 seconds?

Pro, 2017-09-28:
As far as I understood keepinit is the timeout until the empty package is sent to tell the server "I'm still there". So if you set keepinit to e.g. 5 seconds it will cause more network traffic since at least every 5 seconds a packet will be sent. As also mentioned in the linked page above, responding to the keepalive is optional. So it may also happen that the OS on the server side does not respond to the keepalive even if the connection is still up, which would cause the client to close the connection. In general major OSes respond to the keep alive.

I don't think that you can use it in any way to reconnect. The use of the keepalive is that after the specified number of tries the connection is closed and the next time the user code calls send it gets an error. Before that timeout, send just caches the packages up to a specific limit.

So it really depends on your setup how you want to handle that case.
If you have a unstable network and you want to detect broken connections fast and reliable, then it would make sense to reduce the keepinit to a smaller value, or if you want to have it in a reliable way you need to send real TCP messages where you are requesting a response from the server, e.g. read the ServerStatus variable from the server.

Using the variable read you can keep the OS default value for keepinit, because reading variables uses the configured timeout of the source code.

 i.e. do you care about the additional spamming of the network every 5 seconds

WoutvdBerg, 2017-09-28:
Thanks for the clarification, I know everything I need to know now.

As far as I'm concerned the issue can be closed.

Thanks!

jpfr, 2017-09-28:
There are two goals we want to pursue in parallel:
1. Use sensible TCP/IP settings to be a good citizen on the network
2. Have clear, understandable + documented failure modes that users can rely on

What happened here was an **unexpectedly large** timeout. The users didn't expect it. And even we developers didn't expect it. During 75s, the connection was unusable. Even though it could have been recovered by a reconnect. To me, that's a bug.

From http://coryklein.com/tcp/2015/11/25/custom-configuration-of-tcp-socket-keep-alive-timeouts.html, I understood that there are two timeouts for pruning stale connections:

- tcp_keepalive_time: time until the first keepalive ACK (default 7200 seconds)
- tcp_keepalive_intvl: retry timeout for the keepalive ACK (default 75 seconds)

After plugging the cable, the OS tried to reestablish the connection during 75s.
To me, it would be okay to set tcp_keepalive_intvl to a lower value. Even one second would be high enough on todays network. And we won't hit this ery often. The securechannel keepalive is smaller than the 72000 seconds for TCP-keepalive anyway.

I see two options going forward:
- Make the client **always respect the timeout value**. So users know after `timeout` ms whether the service call went through or the connection is broken for good. So the reconnect happens within this time frame.
- Split up the client `timeout` into several values with well-documented meaning and impact on the different failure modes.

And most important: Documentation!!

jpfr, 2017-10-30:
This issue came up on the mailing list: https://groups.google.com/forum/#!topic/open62541/cOfGr2mPnoM

StalderT, 2017-10-30:
can the PR https://github.com/open62541/open62541/pull/1288 affect/resolve this issue?

WoutvdBerg, 2017-11-15:
@Stalder only possible to check if #1273 is fixed.

WoutvdBerg, 2017-11-16:
@StalderT, the reconnecting is working better now but sometimes with errors, used fix #1370 to test
example: 
[clientExample securechannel error.zip](https://github.com/open62541/open62541/files/1477662/clientExample.securechannel.error.zip)

Within the keepinit timeout the connection is reestablished but sometimes fails with free of non allocated memory error. No clue why or when.

This error sometimes comes along but it results in a extra attempt and then connects:
Error receiving the response with status code BadRequestHeaderInvalid.
The BadRequestHeaderInvalid only occurs when 'Sending a request of type 629' is called multiple times without the responses (because the cable is not connected)

Once there was a broken pipe error after keepinit timeout time when I didn't plug the cable back in instead of trying to reconnect. 

In the example the: 
printf("Not connected. Retrying to connect in 1 second\n"); is called only after keep init time and cable isn't plugged back in.
printf("Connection was closed. Reconnecting ...\n"); is called only after cable is plugged back in and starts a new connection.
Seems right if it is intended behavior

WoutvdBerg, 2017-12-04:
@StalderT, any updates? I tried master and reconnecting works if cable is plugged in before some sort of timeout. If I don't plug cable back in within ~5 seconds there is a malloc break.
It doesn't always occur but most of the times it does. Example used same as above.

StalderT, 2017-12-04:
I can't reproduce the error. 

Just to make a test, can you try [welcome.zip](https://github.com/open62541/open62541/files/1526679/welcome.zip) that includes https://github.com/StalderT/open62541/commit/a81e2075bfbe2df8bad3b2d1e18357fac4b1ae77

If it does not work, can you send Wireshark log and client run loglevel 100 with actual open62541 version ?

Do you unplug the network cable on the server or on the client?

Is Wireshark running on the client or on the server?

Does valgrind work on your platform?

StalderT, 2017-12-04:
@WoutvdBerg sorry, please use this one : [welcome.zip](https://github.com/open62541/open62541/files/1526725/welcome.zip)

WoutvdBerg, 2017-12-04:
With given example malloc errors.

Pastebin loglevel 100: https://pastebin.com/raw/rddwSTAS
Wireshark run: 
[reconnecting malloc error.pcapng.zip](https://github.com/open62541/open62541/files/1527353/reconnecting.malloc.error.pcapng.zip)

The malloc error doesn't always occur but occurs when these 2 are after each other:
`[12/04/2017 14:08:57.560] debug/client  Sending a request of type 629
[12/04/2017 14:08:58.560] debug/client  Sending a request of type 471
`

valgrind doesn't work.

Cable unplug on client, also tested on server but both have the same result.

StalderT, 2017-12-04:
How do you compile and start you program ?

[image]

WoutvdBerg, 2017-12-04:
Comiling and running in netbeans with mac clang compiler.
Which server are you using?

StalderT, 2017-12-05:
Can you post your NetBeans project ?

Which version of NetBeans and clang do you use (clang -v)?

WoutvdBerg, 2017-12-05:
Apple LLVM version 9.0.0 (clang-900.0.38)
Netbeans version 8.1
[test_connectionBreak.zip](https://github.com/open62541/open62541/files/1530647/test_connectionBreak.zip)

StalderT, 2017-12-05:
I can't reproduce the error.

Just to have an idea where application crash, can you use [open62541.c](https://github.com/open62541/open62541/files/1531080/open62541.c.txt) and send logs?

WoutvdBerg, 2017-12-06:
I will debug tomorrow in open62541. 
How do you break the connection?
Which server are you using?

StalderT, 2017-12-07:
I use open62541 server. I also tried with opc.tcp://opcua.demo-this.com:51210/UA/SampleServer

To break the connection, I remove the network cable between the server (or my ISP router) and the client.

WoutvdBerg, 2017-12-08:
I tried with an older version of master, implemented and it works good. No errors, except for the badconnection closed almost immediately after cable removal. So the original issue is fixed.

If the other issue reoccurs I will let you know.

closed 2017-12-08 by WoutvdBerg

WoutvdBerg, 2017-12-11:
@StalderT, I think I found the reason for the crashes. Issue #1443 created because this has nothing to do with the timeout.