open62541/open62541 #1231: UA_malloc(0) returning NULL is considered an error.

muhgin, 2017-10-03:
Description
===========
In function ServerNetworkLayerTCP_getJobs particularly, when "layer->mappingsize" is 0, a call to malloc(sizeof(UA_Job) * (size_t)((layer->mappingsSize * 2))) returning NULL is considered an  error, where-else according to specifications,  a call to malloc(0) can return "a null pointer or a unique pointer that can be successfully passed to free()".

Background Information / Reproduction Steps
===========================================
Running UA_Server_run_iterate() for the first time will call ServerNetworkLayerTCP_getJobs() which returning "UA_STATUSCODE_BADOUTOFMEMORY" because of a call to malloc(0).
This return code is not checked in UA_Server_run_iterate(), but rather the value is assigned to variable "jobsSize". This cause segmentation fault in my microcontroller.

Im using ESP32 esp-idf.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.2
 - [x] Other OPC UA SDKs used (client or server):  server
 - [x] Operating system: FreeRTOS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-02-01:
@muhgin is this still an open issue in 0.3?

My guess is that since @Gfast2 managed to get open62541 running on ESP32 the issue is resolved?

In the coming weeks I will also do some ESP32 tests and reopen this issue if necessary.

closed 2018-02-01 by Pro

muhgin, 2018-02-02:
@Pro , I'm not sure about 0.3 but I had 0.2 running for server and client after fixing some(i cant remember how many, it could be just this one) compatibility issues.

I will also test 0.3 if needed in the future.