open62541/open62541 #1240: __UA_Client_readAttribute hides legal behaviour behind synthetic UA_STATUSCODE_BADUNEXPECTEDERROR

miseler, 2017-10-10:
Description
===========
While trying to read the value attribute of a certain node the server returns no value (hasValue==0) but with service result UA_STATUSCODE_GOOD.
My interpretation of the standard specification is that this is legal behaviour (at least for the value attribute). UAExpert displays it as I would expect: StatusCode==Good and Value==Null. My client code got UA_STATUSCODE_BADUNEXPECTEDERROR instead and investigation showed that this error code is synthesized in __UA_Client_readAttribute.
The library should not hide legal behaviour behind synthetic error codes.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): v0.2 (but code looks unchanged in master)
 - [x] Other OPC UA SDKs used (client or server): CODESYS Control for Raspberry Pi SL & UAExpert 1.4.4
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

[image]

[wireshark.ZIP](https://github.com/open62541/open62541/files/1372342/wireshark.ZIP)

jpfr, 2017-10-12:
I added some tests for reading empty variables in #1244.
Looks good on 0.2 and master.

With respect to __UA_Client_readAttribute, this method is wrapped into static inline functions. Such as 
```c
UA_StatusCode
UA_Client_readNodeClassAttribute(UA_Client *client, const UA_NodeId nodeId,
                                 UA_NodeClass *outNodeClass);
```
So the result is directly written into the target variable. In this case the NodeClass (an integer).
We need a way to differentiate the "no variable" returned case. That can only happen via the status code.

Note that `UA_Client_readNodeClassAttribute` is a convenience method with an abbreviated signature. If you use the "native" service `UA_Client_Service_read` you get a DataValue that can have "no value set" as an answer for the optional node attributes.

Do you have another proposal for `UA_Client_readNodeClassAttribute` et al.?

gerdBert, 2018-02-13:
I tested the method 
```cpp
UA_StatusCode
UA_Client_readNodeClassAttribute(UA_Client *client, const UA_NodeId nodeId,
                                 UA_NodeClass *outNodeClass);
```
I was not able to read the NodeClass of several variables. Both server and client are build with Open62541. The methods always returned UA_STATUSCODE_BADUNEXPECTEDERROR. When I dived into `__UA_Client_readAttribute` I found out that `res->value.type == outDataType` leads to `Int32 == NodeClass`.

Pro, 2018-02-14:
@gerdBert which version are you using?
This was fixed in commit  some time ago: https://github.com/open62541/open62541/commit/6ae9c485e126c62254f14f641900706ede072d45

See also #1262

gerdBert, 2018-02-16:
I am using version v0.2. 
In this former version the error was still present.
Good to know that it is already fixed.

Pro, 2018-02-16:
I leave this open as a reminder to backport this to the 0.2.1 maintenance relase

Pro, 2018-02-20:
This is already backportet to the maintenance release:

closed 2018-02-20 by Pro