open62541/open62541 #1242: Client socket is not closed in the network plugin implementation

jonasgreen88, 2017-10-12:
According to the description of the Close connection function pointer in the Connection structure in the network plugin the close function shall only shutdown the socket and let the listen handler close it. But how shall this be handled for a client which doesn't listen for connections? Is there a particular reason to only shutdown, but not close the connection in the close function?
https://github.com/open62541/open62541/blob/fb350446f339c9a4c7cbf36b599c587e6ad54e54/include/ua_plugin_network.h#L104

In the network plugin implementation both the server's accepted connections and a client connection uses the same close function, which only shutdown the socket. It looks like that the client connection is never closed, only shutdown.

jpfr, 2017-10-12:
True.
I suggest we set a different method for the close-callback in the client case.

jpfr, 2017-10-12:
Fixed via #1242.

commit 9b9d7f60 (2017-10-12): shutdown+close client sockets (fix #1242)
closed 2017-10-12 by jpfr