open62541/open62541 #1246: server_nodeset_plc not working

santoshpanda1989, 2017-10-13:
--------------------------------------------------------------------------------

Description
===========
I have built the example nodeset compiler by adding custom xml to it. But the server is exited with an error "Adding the DI namespace failed. Please check the previous output and error". I have attached the Cmakelists.txt in the examples/nodeset folder. Please let me know if I am missing something in order to implement it.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): nodeset_compiler2
 - [ ] Other OPC UA SDKs used (client or server): Prosys
 - [ ] Operating system: OSX
[CMakeLists.txt](https://github.com/open62541/open62541/files/1381823/CMakeLists.txt)

 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-10-13:
I just fixed an error with duplicate node ids in the branch. See
#1238

Can you try that version?

Adding PLCopen should work there. See 
https://github.com/open62541/open62541/blob/cf21ac33ddac1fb965380caf40c187a106fbc1a8/examples/nodeset/CMakeLists.txt#L84

santoshpanda1989, 2017-10-13:
@Pro I tried the updated nodeset_compiler2 version. I have added my custom XML file In the example/node-set/cmakelists.txt. PFA the cmakelists.txt and custom XML. I have added the custom xml instead of opc.ua.nodeset2.plc.xml.
[Archive.zip](https://github.com/open62541/open62541/files/1382325/Archive.zip)

Pro, 2017-10-13:
If you check the output it says:

```
[10/13/2017 11:54:42.166] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Type of object node must be ObjectType and not be abstract
[10/13/2017 11:54:42.166] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
[10/13/2017 11:54:42.166] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference is invalid
[10/13/2017 11:54:42.166] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Parent node not found
...
```

The first message is the most important one. Debugging into the code (setting a breakpoint where this message is printed, shows that the faulty node is `i=5023` with name `1:CFU` (Yes, these messages can be improved a bit, e.g. by showing the corresponding node id. But that's an open todo for now).

If you check your information model, you have the following:

```
i=5001 (1:UNIT, Type Variable) -> Organizes -> i=5023 (1:CFU, Type Object)
```

So you defined a Variable which Organizes an Object. According to the Specification this is not allowed:

[image]
(Source: OPC UA Specification, Part 3 Address Space Model, Page 55)

The source node of Organizes can not be a Variable node, therefore adding your .xml fails.

You need to change your model then it should work.

Please let me know if that fixes your issue and leave any comment or close the issue.

santoshpanda1989, 2017-10-13:
@Pro I saw that issue as soon as I replied. But still, I have the issue as "Parent Node invalid". Could you please give me the idea about how to use the breakpoint to fetch the node ids?

Pro, 2017-10-13:
Did you fix your model already?

The parent node not found from my comment above is the result of the first error:

`Type of object node must be ObjectType and not be abstract` has the result that the node 5023 is not added to the node store.
Following that there is a node which references 5023, doesn't find it and thus you see `Parent Node not found`.
As soon as you fix the first error, the parent node should be found.

The nodeset compiler already checks if all the required parent nodes are within the nodeset and alerts you before even compiling the source. So `Parent node not found` normally is a result of previous errors.

santoshpanda1989, 2017-10-13:
@Pro Apart from the object type error, there was another error present for node id 5023. After solving the error it is working fine. I thank you for your help. Hence closing the issue.

closed 2017-10-13 by santoshpanda1989