open62541/open62541 #1249: How to create a custom DataType

syvaren, 2017-10-13:
Description
===========

I have created a custom datatype with the UA_Server_addDataTypeNode().  The type is a child of UA_NS0ID_ENUMERATION and has a property EnumStrings that is an array of type UA_NS0ID_LOCALIZEDTEXT.

When I create a Variable node with a dataType of my custom datatype, it creates the node, but I get errors trying to set the value of this node.

How do I correctly write a value of my custom datatype to this node?

UA_UInt32 index = 0;
I have tried using UA_Variant_setScalar(&variant, &index, &UA_TYPES[UA_TYPES_UInt32]), but keep getting type mismatches.  I cannot seem to figure out how to set a value of my custom datatype.

Any help is much appreciated.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): .3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Debian 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-10-13:
Did you already check the examples in https://github.com/open62541/open62541/tree/master/examples/custom_datatype ?

syvaren, 2017-10-13:
@Pro 
No, lol, I didn't know there were examples.  I checked for examples in the html documentation and read every entry I could find in the header about DataType though.

I will check these out, Thank you!

syvaren, 2017-10-30:
@Pro I was successful in creating my own DataType, thanks for making me aware of the example.

I do have a question about usage though.  I created my own enumerated type and created a variable that has this type.  Instead of showing my DataType in the type field it shows Int32, also the current value is only a number even though the definition has enumstrings.

Pro, 2017-11-13:
How exactly did you create your own data type?

Can you post some example?

Pro, 2018-02-01:
See also the related discussion here: #1420 and #1414 

DataType is fixed in #1559

closed 2018-02-01 by Pro

juangburgos, 2019-03-12:
> Did you already check the examples in https://github.com/open62541/open62541/tree/master/examples/custom_datatype ?

That example shows how to use `UA_Server_addVariableTypeNode` but not `UA_Server_addDataTypeNode`. Is there any example or somewhere n the docs where we can find how to use `UA_Server_addDataTypeNode`?