open62541/open62541 #1250: Build errors with the latest master commits

syvaren, 2017-10-13:
Description
===========
I am getting the following build errors out of the latest master commits.  Any ideas on what I'm missing?
[image]

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Debian 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-10-13:
We had to add an additional dependency for python.

`pip install six`

See https://stackoverflow.com/questions/21990581/how-to-install-the-six-module-in-python2-7 or other sources, e.g. google for `ImportError: No module named six`

syvaren, 2017-10-13:
@Pro 

Thanks for the fix, there is still the warning newEntry shadowing a global variable that is treated as an error due to -Werror in the flags files.  I removed it to get past it, but is this a known issue?

Pro, 2017-10-13:
Ah interestingly this wasn't detected by the CI/Travis.
Anyways, it's fixed now in master with https://github.com/open62541/open62541/commit/c86c8e833f2e0911f64727a059a87a6f3102bb6b

syvaren, 2017-10-13:
@Pro 
Thanks!

jpfr, 2017-10-14:
Hey @Pro ,

can we add the dependency on python-six to /doc/building.rst?
Otherwise, these kinds of issues will continue to crop up.

syvaren, 2017-10-30:
@Pro I think this is fixed now, I no longer see the shadowing variable error, and I saw a commit that said it added installing six in the guide.

closed 2017-10-30 by syvaren