open62541/open62541 #1251: Not able to build in windows7

santoshpanda1989, 2017-10-16:
Description
===========
I am not able to build in windows 7.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: windows 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-10-16:
Which compiler/Visual Studio are you using?

Which CMake Flags? Did you enable "UA_ENABLE_FULL_NS0"? If yes, you can try to set the UA_ENABLE_FULL_NS0=OFF

VS2008 does not support strings longer than 65535 bytes, but that should already be covered here:
https://github.com/open62541/open62541/blob/c205c4d5455a66a27dab32f9a303db4908139475/CMakeLists.txt#L487

Can you try to tweak that line and make sure `NODESET_MAX_STR_LEN` is set to 65535

santoshpanda1989, 2017-10-16:
@Pro  The compiler is msvc 14.10 and visual studio version is 2015. When I tried with UA_ENABLE_FULL_NS0=OFF then there isn't any error. The NODESET_MAX_STR_LEN is set to 65535.

Pro, 2017-10-16:
Ok, so if you set `UA_ENABLE_FULL_NS0=ON` and do not modify `NODESET_MAX_STR_LEN` it is set to 65535? It should be `0` for VS2015.

I just saw that we do not yet test  `UA_ENABLE_FULL_NS0=ON` on appveyor. I should add that to the `appveyor.yml` so that this is also tested.

Anyways setting `NODESET_MAX_STR_LEN` will have the effect that the value of various nodes, especially `OpcUa_BinarySchema` (i=7617)  will not be set since it's longer than 65535 bytes.

santoshpanda1989, 2017-10-16:
@Pro PFA the error log from Visual Studio. I also tried to set the NODESET_MAX_STR_LEN to 0 but still I got the errors.
[Error.txt](https://github.com/open62541/open62541/files/1387319/Error.txt)

Pro, 2017-10-16:
Thanks, if you currently need the full NS0 and you want to compile, you can manually set `NODESET_MAX_STR_LEN=65535` in you CMakeLists.txt until this is fixed.

santoshpanda1989, 2017-10-16:
@Pro I did that but the executables generated in build/bin/examples/debug are not functioning.

Pro, 2017-10-16:
After setting `NODESET_MAX_STR_LEN=65535` e.g. here:
https://github.com/open62541/open62541/blob/c205c4d5455a66a27dab32f9a303db4908139475/CMakeLists.txt#L486
and removing the `if ...` afterwards make sure that you clean your build directory (i.e. re-run Cmake).

This should definitely generate a `ua_namespace0.c` file which does not contain any string longer than 65535 bytes.

What does `not functioning` mean? Please provide any debug log an error output.

santoshpanda1989, 2017-10-16:
@Pro I re-ran the whole process. There wasn´t any error. However when I executed the examples, it closed unexpectedly. PFA the debug report of the same.
[server_nodeset_plcopen_debug.txt](https://github.com/open62541/open62541/files/1387441/server_nodeset_plcopen_debug.txt)

Pro, 2017-10-16:
Can you build with Debug or start the example with debug so that we can see where exactly it fails.

Just to be sure:
For this test your are using the original code from the master branch without any modifications on your side?

santoshpanda1989, 2017-10-16:
@Pro Yes, I am using the master branch without any modification. PFA the screenshot of debug.
[image]

Pro, 2017-10-16:
I just ran the example on my PC (VS 2015) with `NODESET_MAX_STR_LEN=65535`  and got a more useful error message, namely 

> Ausnahmefehler bei 0x008D58A9 in server_nodeset_plcopen.exe: 0xC00000FD: Stack overflow (Parameter: 0x00000000, 0x001D2000)

So the problem here is that the `ua_nodeset0.c` file is quite large and the stack for the program is too small.
It worked for me after I increased the stack size to 10MB:

Project (`server_nodeset_plcopen`) -> Properties -> Configuration Properties -> Linker -> System -> Stack Reserve Size = 10 000 000

Does this also work for you?

santoshpanda1989, 2017-10-16:
@Pro Thanks for the help. Now it is working fine. Please let me know when NODESET_MAX_STR_LEN=0 in the Cmakelists.txt can be added.

Pro, 2017-10-16:
To be able to use strings longer than 65535 bytes, we need to use static char arrays with the corresponding ascii number instead of a real string. This requires some coding in the nodeset compiler, see #1254

Additionally to fix the stack size issue, we probably need to split the huge nodeset compiler output into multiple files. See #1255

I will close this issue in favour of the other two.

closed 2017-10-16 by Pro