open62541/open62541 #1255: Split huge generated files of nodeset compiler into multiple files

Pro, 2017-10-16:
This is an issue which can be grabbed by first time contributors and is fairly easy to implement.

The nodeset compiler just generates one huge file with one huge function which initialized the address space.

To avoid issues with stack overflow on Windows (e.g. see #1251) this file should be split into multiple subfiles and sub functions.

jpfr, 2017-10-16:
Hey @Pro 
Do we have to split files? Our amalgamation is huge and compiles reasonably fast.
Maybe it's enough to split the namespace generation into smaller methods.

Pro, 2017-10-16:
I'm not sure if the stack overflow in Visual Studio (as mentioned in #1251) is caused by the huge method or huge file.
Probably it's the huge method, so first try should be to split the method into sub-methods (which also makes the whole process easier to handle compared to multiple files)

Pro, 2019-05-22:
This was fixed with #1424. The huge method is now split into one method per node

closed 2019-05-22 by Pro