open62541/open62541 #1257: Additional nodes with the nodeset compiler

santoshpanda1989, 2017-10-17:
Description
===========
I built and ran the server_nodeset example with the server_nodeset.xml. In the example XML file there are two variables but in the client application, I can see an additional variable.  PFA the screenshot
[image]
 of the same.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): Prosys OPC UA Client
 - [ ] Operating system: OSX
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

santoshpanda1989, 2017-10-17:
Is it a mistake from my end or is it a genuine issue? I searched for the extra NodeId in the generated code but could not find it.

Pro, 2017-10-17:
I think this is desired behaviour @jpfr ?

In the .xml there is the following (simplified) structure:
```
testType (ObjectType, i=1001) -> HasComponent -> Var1 (Variable, i=6001)

testInstance (Object, i=5001) -> HasTypeDefinition -> testType
testInstance (Object, i=5001) -> HasComponent -> Var1 (Variable, i=6002)
```

So this means when the testInstance is created, the Var1 (i=6001) is instanciated from the type as a mandatory child.
Then an additional variable is added as Var1 (i=6002).

Maybe we should change the name of i=6002 to Var2?

Pro, 2017-11-09:
I think this issue can be closed since it is intended behaviour.

Anyways I changed the name of the second Var1 to Var2.
See
https://github.com/open62541/open62541/commit/add1b806ac173ca4bd0a26746be460617803a2df

closed 2017-11-09 by Pro