open62541/open62541 #1258: Including private plugin header file in sources

jonasgreen88, 2017-10-17:
Hi,

In ua_server_discovery.c a private header file of the plugin implementation is included. When I integrate the stack to my project I have developed my own config plugin and it is not named according to the example in this repository, and I don't think it has to  be.
https://github.com/open62541/open62541/blob/a894e31e415bf469f69ff2f3b26da32debbffd59/src/server/ua_server_discovery.c#L7

In my opinion the source files of the stack should only include private header files of the source and public header file of the include/ dir. My proposal is to add a callback function that is declared in a public header file of the include/ dir and implemented in the config plugin by the user. In ua_server_discovery.c this function is called to retrieve the client config when needed. What do you guys think?

Pro, 2017-10-17:
Thanks, this would definitely be an improvement.
Maybe you can create a PR and we can see how much it changes?

jonasgreen88, 2017-10-17:
Ok, I can do that. I just wanted to bring up the issue first to see if you agree to the change. I'll look into it during the coming days.

Pro, 2018-02-01:
Closing in favor of #1264

closed 2018-02-01 by Pro