open62541/open62541 #1261: Limited memory resources on embedded device

jonasgreen88, 2017-10-17:
Description
===========
We run this stack on an embedded device and have ported the memory management api of the stack to our heap implementation that has a set of fixed sized memory partitions and a fixed number of memory blocks per partition size defined.  Now when we run the UA Compliance Test Tool we see problems where the test sends requests to our server implementation which results in responses that are larger than our heap can manage. As the heap returns a null pointer the stack generates a Bad_OutOfMemory response back to the CTT which it doesn't accept. As a client can choose to read many nodes, references etc. in a single request there must be a way for the server to respond with an error code telling the client that the total response size is too large to handle, especially for an embedded device with limited memory resources.

A concrete example is a test case (View Services -> View RegisterNodes -> Test cases -> Err-004) where the CTT sends a RegisterNodes request including 500 node ids. The request allocates a ~10K buffer and the response needs another ~10K buffer when mirroring the node ids of the request to the response. The stack implementation forwards the UA_BadOutOfMemory status code if no memory is available in the memory heap. But the test case expect either Good or Bad_TooManyOperations as response status. In this case I think the stack should somehow have knowledge about its resource limitations and return Bad_TooManyOperations. When receiving the Bad_TooManyOperations status a client could then re-send the request with less operations included.

We are also struggling with a similar issue where we have an object node in our address space that references many variable nodes. When a client reads the references of the object node the response buffer needs 120 bytes per reference. The required size of the response buffer grows quickly when the number of variable nodes are increased. Right now we have to limit the number of variable nodes that can be referenced from the object node depending on the largest heap memory partition. In some way we would like to respond with a status code to tell the client to read a subset of the references per request to allow our device to process the request with its limited memory resources.

To get arround these issues for now we have just increased the memory heap with larger parititions and the amount of blocks per partition when needed. But this just solves the requests that the CTT sends, some other client could potentially send requests where the response requires even larger buffers and then we don't wan't our device to return Bad_OutOfMemory as this doesn't seems to be the correct behavior according to the CTT.

Have you guys thought about these kind of issues?

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 5ec8dd51ec4410b995fdac93eac230e0165dbcbb
 - [ ] Other OPC UA SDKs used (client or server):  UA CTT 1.02
 - [ ] Operating system: Proprietary embedded OS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2017-10-17:
If you have problems allocating memory for the answer, you could just limit max message size and number of chunks in the network layer config.

https://github.com/open62541/open62541/blob/master/plugins/ua_config_default.c#L39

The client should notice these settings during the handshake.

Still the error status should be fixed ;)

Pro, 2017-10-17:
An additional note, even if we do not (yet) implement it, OPC UA Specification Part 5 defines additional nodes which can be set by the server to indicate resource limits:

[image]

It is up to the developer to set these limits and then the client should respect those.
Maybe we/you/whoever wants to code some lines, can add some simple checks if these limits are respected by the client, and if not return `Bad_TooManyOperations`.

A major issue here is, how does a C program know how much memory is left, and when it knows the amount of bytes, how do you know how much space the call will take. Only if you know all that numbers you can handle that.

Another fairly simple approach would be to return `Bad_TooManyOperations`  as soon as the server runs out of memory. But does that really help the client to understand the error? IMHO `BadOutOfMemory` gives me exactly the information I need as a client, i.e., the server has no memory left.

It could also happen that RegisterNodes takes up almost all the memory, but leaves some bytes free, and succeeds. The next call to RegisterNodes tries to register one node, which will fail due to OutOfMemory. What should be returned then? OutOfMemory or TooManyOperations?

(Just my opinion on this issue. Since I never really worked on such hardware limited devices I'm definitely not an expert on it)

HolgerJeromin, 2017-10-17:
Accepting Bad_TooManyOperations but not Bad_OutOfMemory looks like a bug (or at least strange thing) in the ctt.

Stasik0, 2017-10-18:
Can someone open a mantis ticket?

jonasgreen88, 2017-10-18:
@Stasik0 The MaxMessageSize limits the maximum request message size the client is allowed to send to the server. It doesn't put any constraints on the response message size. Similar MaxChunkCount also just limits the max number of chunks a request is allowed to have.
I can put a mantis issue on the CTT so we can find out if this is an issue of the CTT.

jonasgreen88, 2017-10-18:
@Pro I would say Bad_OutOfMemory would be an accepted error code if the client tries to create a resource on the server, for example creating a node. Deleting another node to free up memory could potentially result in that creating the node succeeds when redoing the request. But in our case, where the client sends a request which results in a response that is larger than our memory heap can ever handle, Bad_TooManyOperations gives the client a better clue what to do to make it possible for the server to fulfill the request -> reduce the number of operations which will lead to a lower response size which the server might be able to handle.

jpfr, 2017-10-18:
@jonasgreen88 To summarize:
- If allocating the "results" array fails, we return Bad_TooManyOperations.
- If we have an error within an operation, we continue to return Bad_OutOfMemory.

Creating the results array and looping over the operations has been generalized in a reusable method.
It could be enough to just change the error code in this line:
https://github.com/open62541/open62541/blob/master/src/server/ua_server_utils.c#L348

closed 2023-09-18 by jonasgreen88