open62541/open62541 #1263: small memory leak on  UA_decodeBinary

clepelli, 2017-10-18:
Description
===========

DrMemory analysis reports a leak when I use the UA_DataValueCopy : 

Error 9: LEAK 8 direct bytes 0x0000000006ca0500-0x0000000006ca0508 + 0 indirect bytes
0 replace_calloc                       [d:\drmemory_package\common\alloc_replace.c:2616]
1 UA_decodeBinary                      [.../open62541.c:7323]
2 UA_NodeId_decodeBinary               [.../open62541.c:1026]
3 processServiceResponse               [.../open62541.c:25388]
4 UA_Array_delete                      [.../open62541.c:5793]
5 UA_Array_copy                        [.../open62541.c:5760]
6 Variant_copy                         [.../open62541.c:5124]
7 DataValue_copy                       [.../open62541.c:5509]
8 copy_noInit                          [.../open62541.c:5634]
9 UA_copy                              [.../open62541.c:5659]
10 UA_DataValue_copy                    [.../open62541.h:5779]
11 ...

```
UA_StatusCode
UA_decodeBinary(const UA_ByteString *src, size_t *offset,
                void *dst, const UA_DataType *type) {
    /* Initialize the destination */
    memset(dst, 0, type->memSize);

    /* Set the (thread-local) position and end pointers to save function
       arguments */
    pos = &src->data[*offset];
    end = &src->data[src->length];

    /* Decode */
    UA_StatusCode retval = UA_decodeBinaryInternal(dst, type);

    /* Clean up */
    if(retval == UA_STATUSCODE_GOOD)
        *offset = (size_t)(pos - src->data) / sizeof(UA_Byte); << HERE
    else
        UA_deleteMembers(dst, type);
    return retval;
}
```

Background Information / Reproduction Steps
===========================================
I have not investigated much, but maybe a simple off by one.

Related question : I copy the UA_DataValue in a subscription callback (registered via UA_Client_Subscriptions_addMonitoredItem) to pass it on another thread, which will delete it once processed, because it seems that the UA_DataValue provided to the callback is freed automatically by the library, is it correct ? I was not able to find any documentation on this topic. 

Best regards !

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): Git-Revision: v0.2-45-gc9645d5
 - [x] Other OPC UA SDKs used (client or server): N/A
 - [x] Operating system: windows 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-10-18:
It could be that this was already fixed in current master, since there were a few bugfixes in the last weeks for memory leaks in the _decodeBinary methods.

Do you have the possibility to test your code with current master?

clepelli, 2017-10-18:
I have tried with current master, there seem to be breaking changes that I cannot take time to investigate : 
- the UA_ClientConfig_standard and UA_SubscriptionSettings_standard have disappeared, they must be renamed to *_default I think ?
- applying this change and without changing anything else, the client_connect function times out. Server is the same & available, it is the prosys simulation server.

Maybe you could point me to a minimal git revision where the fix is ?

But I can confirm there is a real memory leak at each iteration.

Pro, 2017-10-18:
maybe this one:
https://github.com/open62541/open62541/commit/c1632f56723b61a38fd5c01b3bb66f8df3669d2d

or this:
https://github.com/open62541/open62541/commit/aa340f25c2979b10a12c92f257ced0772217f8cb

If this doesn't fix it, can you maybe provide a minimal code sample which allows to reproduce the error?

jpfr, 2017-10-18:
The changes to encoding on master should be backwards compatible.
I'll backport them to v0.2 for v0.2.1.

jpfr, 2017-10-19:
See #1267

jpfr, 2017-10-19:
#1267 was merged.
Can you test with the current 0.2 branch?

clepelli, 2017-10-20:
Sorry I was not able to test before, and thanks for your time. 

Unfortunately, the issue is still there on Git-Revision: v0.2-54-g522620d :

0 replace_calloc                       [d:\drmemory_package\common\alloc_replace.c:2616]
1 UA_decodeBinary                      [.../open62541.c:7316]
2 UA_NodeId_decodeBinary               [.../open62541.c:1040]
3 processServiceResponse               [.../open62541.c:25381]
4 UA_Array_delete                      [.../open62541.c:5807]
5 UA_Array_copy                        [.../open62541.c:5774]
6 Variant_copy                         [.../open62541.c:5138]
7 DataValue_copy                       [.../open62541.c:5523]
8 copy_noInit                          [.../open62541.c:5648]
9 UA_copy                              [.../open62541.c:5673]
10 UA_DataValue_copy                    [.../open62541.h:5778]

jpfr, 2017-10-20:
Hello @clepelli ,

1. Yes, the client SDK deletes the notification content after the callback.
2. I misunderstood the trace from drmemory. I thought the memory was lost during decoding.
Now my guess is that the memory is allocated during decoding. But the pointer is "lost" only later.

I still have questions regarding the trace:
```
2 UA_NodeId_decodeBinary [.../open62541.c:1040]
3 processServiceResponse [.../open62541.c:25381]
4 UA_Array_delete [.../open62541.c:5807]
```
Is this a call stack? I don't think that UA_NodeId_decodeBinary or UA_Array_delete ever call processServiceResponse...

By the way, do you edit the `value` within the `processServiceResponse`?

@open62541/open62541-core 
I propose we make the value const within the callback. That's consistent with the other callback mechanism that forward data to the user.

clepelli, 2017-10-20:
Hello,

I do not understand neither this trace from drmemory. The tool instruments the memory allocations and tells at the end of run which of them have not been freed. So the trace shown is usually where the allocation is + sometimes where the loss of the last reference is. Here it seems to be mixed...

I don't edit the value, in the callback, I just deep copy it for later usage.
Here is my whole callback fn : 
```
void opcua_notification_cb(UA_UInt32 monId, UA_DataValue *value, void * context) {

    auto ua_data_value_copy = new UA_DataValue();
    UA_DataValue_copy(value, ua_data_value_copy);
    auto updates_list = ((vector<pair<UA_UInt32, UA_DataValue *>> *) context);
    updates_list->emplace_back(monId, ua_data_value_copy);

}
```

It is called in a dedicated thread looping on manuallySendPublishRequest, here is the whole thread function : 
```    
void opcua_notifications_slave(string const& server_address, bool const authenticate, string const& username, string const& password, UA_UInt16 const root_ns_index,
                               unordered_map<string, UA_ReferenceDescription> const& monitored_references)
{

    auto client = UA_Client_new(UA_ClientConfig_standard);

    UA_StatusCode retval;
    if(authenticate) {
        retval = UA_Client_connect_username(client, server_address.c_str(), username.c_str(), password.c_str());
    } else {
        retval = UA_Client_connect(client, server_address.c_str());
    }

    opcua_notifications_slave_exit = false;
    zsock_t * opcua_notif_socket = zsock_new_push("inproc://opcua_notif");

    /* Create a subscription */
    UA_UInt32 subId = 0;
    UA_SubscriptionSettings settings = UA_SubscriptionSettings_standard;
    settings.requestedPublishingInterval = 50;
    UA_Client_Subscriptions_new(client, settings, &subId);

    /* Add the monitored items */
    vector<pair<UA_UInt32, UA_DataValue *>> updates_list(0);
    unordered_map<UA_UInt32, string> monitoring_id_to_var_identifier;
    int i = 0;
    for(auto const& ref : monitored_references) {
        UA_UInt32 monId = 0;
        UA_Client_Subscriptions_addMonitoredItem(client, subId, ref.second.nodeId.nodeId, UA_ATTRIBUTEID_VALUE,
                                                 &opcua_notification_cb, &updates_list, &monId);
        monitoring_id_to_var_identifier.emplace(monId, ref.first);
        if (monId) {
            spdlog::get(LOGGER).get()->info("Monitoring variable {}, id = {:d}", ref.first.c_str(), monId);
        }
        i++;
    }

    do
    {
        updates_list.clear();
        UA_Client_Subscriptions_manuallySendPublishRequest(client);
        if(!updates_list.empty()) {
            auto values = new vector<pair<UA_DataValue const * const, string const * const>>();
            for(auto const& update : updates_list) {
                auto& identifier = monitoring_id_to_var_identifier.at(update.first);
                values->emplace_back(update.second, &identifier);
            }

            auto evt = new OPCUASubscriptionEvent(values);
            send_ptr(opcua_notif_socket, (intptr_t) evt);
        }
    } while(!opcua_notifications_slave_exit);

    zsock_destroy(&opcua_notif_socket);

    /* Delete the subscription */
    if(!UA_Client_Subscriptions_remove(client, subId)) {
        spdlog::get(LOGGER).get()->info("Subscription {:d} removed", subId);
    }

    retval = UA_Client_disconnect(client);
    UA_Client_delete(client);

}
```

Pro, 2018-02-01:
@clepelli did you already test if this is still an open issue on 0.3, if not then we could close this issue?

jpfr, 2019-03-18:
We have no reports from oss-fuzz.
Seems to be fixed for a long time...

closed 2019-03-18 by jpfr