open62541/open62541 #1265: Error: `UA_STRING_NULL' can't be equated to common symbol 'UA_BYTESTRING_NULL'

vladimiroltean, 2017-10-18:
Hi,

Using arm-buildroot-linux-musleabihf-gcc and your v0.2 release tag, compiling `src/ua_types.c` gives me the following error:
```
[ 87%] Building C object CMakeFiles/open62541-object.dir/open62541.c.o
{standard input}: Assembler messages:
{standard input}:32673: Error: `UA_STRING_NULL' can't be equated to common symbol 'UA_BYTESTRING_NULL'
CMakeFiles/open62541-object.dir/build.make:203: recipe for target 'CMakeFiles/open62541-object.dir/open62541.c.o' failed
make[4]: *** [CMakeFiles/open62541-object.dir/open62541.c.o] Error 1
```
I traced this down to the use of the gcc flag '-fmerge-all-constants' in CMakeLists.txt.
I am able to fix the compilation error with this patch:
```
diff -u -r a/CMakeLists.txt b/CMakeLists.txt
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -136,7 +136,7 @@
     # Strip release builds
     if(CMAKE_BUILD_TYPE STREQUAL "MinSizeRel" OR CMAKE_BUILD_TYPE STREQUAL "Release")
         add_definitions(-ffunction-sections -fdata-sections -fno-stack-protector -fno-unwind-tables
-                        -fno-asynchronous-unwind-tables -fno-math-errno -fmerge-all-constants -fno-ident)
+                        -fno-asynchronous-unwind-tables -fno-math-errno -fno-ident)
         set(CMAKE_C_LINK_FLAGS "${CMAKE_C_LINK_FLAGS} -s")
         set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS} -s")
         if(APPLE)
```
Please comment on possibly integrating this patch into your source tree.
Thanks,
Vladimir

Pro, 2018-02-01:
Doc from gcc: (https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html)

> -fmerge-all-constants
> 
> Attempt to merge identical constants and identical variables.
> 
> This option implies -fmerge-constants. In addition to -fmerge-constants this considers e.g. even constant initialized arrays or initialized constant variables with integral or floating-point types. Languages like C or C++ require each variable, including multiple instances of the same variable in recursive calls, to have distinct locations, so using this option results in non-conforming behavior.

> -fmerge-constants
> 
> Attempt to merge identical constants (string constants and floating-point constants) across compilation units.
> 
> This option is the default for optimized compilation if the assembler and linker support it. Use -fno-merge-constants to inhibit this behavior.
> 
> Enabled at levels -O, -O2, -O3, -Os.

@open62541/open62541-core any comments on why to keep or remove `-fmerge-all-constants`?

Pro, 2018-11-29:
As discussed and voted in the telco, we will remove the flag, PR will follow.

PR #2255 (2018-11-29): Removes `merge-all-constants` flag
Fixes #1265
commit 3a836bda (2018-11-29): Removes `merge-all-constants` flag. Fixes #1265
commit 2fcde93f (2018-11-30): Removes `merge-all-constants` flag. Fixes #1265
closed 2018-11-29 by jpfr