open62541/open62541 #1266: Retrieving string from server. Simple issue for experienced users.

alxhoff, 2017-10-18:
!ATTENTION!
Please read the following page carefully and provide us with all the
information requested:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Use Github Markdown to format your text:
https://help.github.com/articles/basic-writing-and-formatting-syntax/

Fill out the sections and checklist below (add text at the end of each line).

Remove this text between the ATTENTION marks after reading it.
!ATTENTION!
--------------------------------------------------------------------------------

Description
===========
Retrieving string from server. 

Background Information / Reproduction Steps
===========================================
This is probably a basic question but I haven't been able to find anything addressing it anywhere. 

I just started using open62541 today so I was just making a basic server client demo where i could set and read a string and int from bother the server and client. Setting and getting the int works fine on bother the client and server but I cannot get the string from either the server or client. I am getting a "Unknown" status code when I use UA_Server_readValue.

The I am using is as follows :
-(creating attribute with index and id_string with a random default value string (1, test.str)
  78     UA_VariableAttributes attr;
  79     UA_VariableAttributes_init(&attr);
  80 
  81     UA_String my_string = UA_STRING(default_val);
  82 
  83     UA_Variant_setScalar(&attr.value, &my_string, &UA_TYPES[UA_TYPES_STRING]);
  84 
  85     attr.description = UA_LOCALIZEDTEXT("en_US", description);
  86     attr.displayName = UA_LOCALIZEDTEXT("en_US", displayname);
  87     attr.dataType = UA_TYPES[UA_TYPES_STRING].typeId;
  88 
  89     UA_NodeId my_string_node_id = UA_NODEID_STRING(index, ID_string);
  90     UA_QualifiedName my_string_name = UA_QUALIFIEDNAME(1, displayname);
  91     UA_NodeId parent_node_id = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
  92     UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
  93     UA_Server_addVariableNode(server, my_string_node_id, parent_node_id,
  94             parentReferenceNodeId, my_string_name, UA_NODEID_NULL, attr, NULL, NULL);

my get function is as follow:

 116 static char* get_string_attribute(UA_Server* server, uint16_t namespace_index,
 117         char* attribute_string_ID)
 118 {
 119     UA_NodeId node_to_get = UA_NODEID_STRING(namespace_index, attribute_string_ID);
 120 
 121     UA_Variant ret_val;
 122     UA_StatusCode retval = UA_Server_readValue(server, node_to_get, &ret_val);
 123 
 124     if(retval == UA_STATUSCODE_GOOD){
 125         printf("String read sucesful with code: %s\n", UA_StatusCode_name(retval));
 126         printf("Read integer with value: %s\n", (UA_Byte*)ret_val.data);
 127         return (UA_Byte*)ret_val.data;
 128     }
 129     else{
 130         printf("String read failed with code: %s\n", UA_StatusCode_name(retval));
 131         return 0;
 132     }
 133 }

It is always failing at line 124.

Any help would be appreciated, I have been through the documentation but cannot find anywhere were a string attribute is retrieved.

Cheers

Checklist
=========
Please provide the following information:

 - [2.0 ] open62541 Version (release number or git tag):
 - [ none] Other OPC UA SDKs used (client or server): 
 - [ Arch Linux] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [[test.zip](https://github.com/open62541/open62541/files/1395189/test.zip) ] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2017-10-19:
is it working with UA_Expert?

just a notice 
> UA_QualifiedName my_string_name = UA_QUALIFIEDNAME(1, displayname);
here 1 is hard-coded

alxhoff, 2017-10-19:
Thanks for the quick reply Sten.

I haven't tried with UA_expert yet but your comment about teh 1 being
hard-coded did the trick.

Big thanks for a nudge in the right direction! much appreciated