open62541/open62541 #1269: check_client_async random failure

Infinity95, 2017-10-20:
Description
===========
The test check_client_async sometimes randomly fails.

Background Information / Reproduction Steps
===========================================
Check out current master and run check_client_async maybe ten times.
One of the runs should produce the error.

> Running suite(s): Client
> [01/01/1601 00:00:00.000] info/network	TCP network layer listening on opc.tcp://infinity-tp:4840
> [01/01/1601 00:00:00.000] info/network	Connection 7 | New connection over TCP from 127.0.0.1
> [01/01/1601 00:00:00.000] info/channel	Creating a new SecureChannel
> [01/01/1601 00:00:00.000] info/channel	Connection 7 | SecureChannel 1 | Opened SecureChannel
> [01/01/1601 00:00:00.000] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
> [01/01/1601 00:00:00.000] info/session	Connection 7 | SecureChannel 1 | Session e4c14788-6516-379c-bb3b-b3abe03e23d2 | ActivateSession: Session activated
> [01/01/1601 00:00:00.000] info/client	Could not decodee the response with Id 5
> [01/01/1601 00:00:00.000] info/client	Error receiving the response with status code BadDecodingError
> Segmentation fault (core dumped)

[failure.pcapng.zip](https://github.com/open62541/open62541/files/1402057/failure.pcapng.zip)

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Ubuntu 16.04 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2017-10-20:
While trying to debug the client i interestingly ran into another different situation:

> [01/01/1601 00:00:00.000] info/network	TCP network layer listening on opc.tcp://infinity-tp:4840
> [01/01/1601 00:00:00.000] info/network	Connection 7 | New connection over TCP from 127.0.0.1
> [01/01/1601 00:00:00.000] info/channel	Creating a new SecureChannel
> [01/01/1601 00:00:00.000] info/channel	Connection 7 | SecureChannel 1 | Opened SecureChannel
> [01/01/1601 00:00:00.000] info/session	Try to use Session with token 00000000-0000-0000-0000-000000000000 but is not found
> [01/01/1601 00:00:00.000] info/session	Connection 7 | SecureChannel 1 | Session e4c14788-6516-379c-bb3b-b3abe03e23d2 | ActivateSession: Session activated
> [01/01/1601 00:00:00.000] info/channel	Connection 7 | SecureChannel 1 | Could not send the message over the SecureChannel with StatusCode BadEncodingLimitsExceeded
> [01/01/1601 00:00:00.000] info/network	Connection 7 | Processing the message failed with error BadEncodingLimitsExceeded
> [01/01/1601 00:00:00.000] info/session	Connection 7 | SecureChannel 1 | Session e4c14788-6516-379c-bb3b-b3abe03e23d2 | CloseSession
> [01/01/1601 00:00:00.000] info/network	Connection 7 | Closed by the server
> [01/01/1601 00:00:00.000] info/network	Shutting down the TCP network layer

This time i don't have a pcap and the server encountered an error. Although this could still be a client error triggered by incorrect data.

jpfr, 2017-10-22:
Hmm. Interesting.
I get `BadTcpMessageTooLarge` in 1 out of 10 tries.
With valgrind activated, all works fine. Must be a timing issue.

This is a blocker for 0.3...

StalderT, 2017-10-24:
Seems to block some PR like https://travis-ci.org/open62541/open62541/jobs/288527026

jpfr, 2017-10-30:
Fixed via #1279.

closed 2017-10-30 by jpfr