open62541/open62541 #1273: V0.2 -> V0.3 BadTcpMessageType

WoutvdBerg, 2017-10-24:
Description
===========
Upgraded my client from V0.2 to V0.3, both amalgamated versions. Changed UA_ClientConfig_Standard to UA_ClientConfig_Default.
Changing back to open62541 .h/.c 0.2 files and _standard the connection is fine. 

Background Information / Reproduction Steps
===========================================

[v0.3 bad tcp type.pcapng.zip](https://github.com/open62541/open62541/files/1411190/v0.3.bad.tcp.type.pcapng.zip)

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2017-10-24:
This is probably the same problam as in #1269

wmjberg, 2017-10-24:
As far as I can see, the other issue is with bad decoding error. This one is just about Tcp message type invalid. The only code executed is establishing connection and after that checking if servertime can be read if not reconnect, if it's good then wait and recheck connection.

WoutvdBerg, 2017-10-26:
Problem now changed to a secure channel id missing sometimes in a readRequest.

Infinity95, 2017-10-27:
#1269 was fixed. Can you check if your issue still persists?

WoutvdBerg, 2017-10-30:
@Infinity95 will check as soon as possible but first have to solve some compiler problems

WoutvdBerg, 2017-11-10:
No more errors, issue probably solved with #1269.

WoutvdBerg, 2017-11-15:
Tried again and error occurs again. Grabbed the client reconnect example with current master but fails on UA_Client_Connect, pointer being freed was not allocated.

jpfr, 2017-11-15:
There is something going wrong on the client side.
The last ReadRequest is sent with a SecureChannelId of zero.

There are two options:
- The client resets the securechannel all by himself
- The memory is written into from the outside

How can we pinpoint which of these cases it is?
Can you provide client logs with debug logging enabled?

WoutvdBerg, 2017-11-15:
https://pastebin.com/raw/yjBD1pXZ
Loglevel 100

It looks like the client resets itself, it's happening everytime in the same order. 1 valueattribute server time comes trough and the next fails. 
Same happens on different servers.

StalderT, 2017-11-15:
Can you provide a minimalist client source code, who generate this issue ?

WoutvdBerg, 2017-11-16:
[clientExample securechannel error.zip](https://github.com/open62541/open62541/files/1477662/clientExample.securechannel.error.zip)

Both the build and the example are in the zip.

WoutvdBerg, 2017-11-16:
@StalderT, texted your fix. Works perfect.

jpfr, 2017-11-20:
Hello there,

We now have a set of unit tests for securechannels where we manually control the timings.
https://github.com/open62541/open62541/blob/d62fdd9641f08e62f182993c70b540d0dd5090f8/tests/client/check_client_securechannel.c#L97
Is it possible to reproduce this bug in such a unit test? That would help in greatly in fixing.
And in ensuring that no regressions happen in the future.

Thanks.

WoutvdBerg, 2017-11-20:
@jpfr, #1370 fixed the problem for me, do I still need to test?

Infinity95, 2017-11-20:
The goal is to reproduce the bug in a unit test, so that it doesn't appear again. Or if it does we notice right away because of the test.

WoutvdBerg, 2017-11-21:
I copied the test to a empty project and tried but most of the times a malloc errors occurs, sometimes it succeeds. 

I added the wireshark logs but I think that's okay because both connect and read return statuscode good

https://pastebin.com/raw/YttiCT7M

[malloc error.pcapng.zip](https://github.com/open62541/open62541/files/1490453/malloc.error.pcapng.zip)

failed log:
https://pastebin.com/raw/behZxa7k

succ. log:
https://pastebin.com/raw/cC8VZV4Y

jpfr, 2017-11-28:
@StalderT's patch is now on master.
Please reopen if required.

closed 2017-11-28 by jpfr