open62541/open62541 #1276: Connecting with FreeOPCUA / Handling of requestHandle

jbvoelker, 2017-10-25:
[open62541.txt](https://github.com/open62541/open62541/files/1414737/open62541.txt)
--------------------------------------------------------------------------------

Description
===========
When running an Open62541 server and trying to connect using a FreeOPCUA based client, FreeOPCUA starts connecting but aborts with following message:
no callback found for message id: ns=0;i=449;, handle: 0

Background Information / Reproduction Steps
===========================================
In Wireshark, everything looks fine until the OpenSecureChannelResponse is sent by the Open62541 server.
The OpenSecureChannelResponse has a requestHandle of 0 in the response header which differs from the requestHandle 1 which is used in FreeOPCUA's request header.
This mismatch causes the FreeOPCUA client to abort the connection attempt.

When applying the attached patch, everything works as expected.
src/server/ua_securechannel_manager.c seems to contain more cases where the request handle is left at zero, maybe this affects other functions I have not tested yet.

In the specification (Part 4, Table 165, Page 140), the requestHandle is defined as follows: "A requestHandle associated with the request. This client defined handle can be used to cancel the request. It is also returned in the response.", so FreeOPCUA seems to be right in relying on the same value being returned.

Steps to reproduce:
- Create an Open62541 server
- Create a FreeOPCUA client
- Start the Open62541 server
- Try connecting to the server using the FreeOPCUA client

Checklist
=========
Please provide the following information:

 - [X] open62541 Version (release number or git tag): master (06dbd3fc3630d9a182d63450df537e71ae414b4d)
 - [X] Other OPC UA SDKs used (client or server): FreeOPCUA
 - [X] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-10-27:
The fix is merged as soon as #1280 goes green.

jpfr, 2017-10-27:
Thanks for reporting.
Can you confirm the fix?

jbvoelker, 2017-10-27:
I have just built the current master, everything works as expected now.
Thank you for fixing the issue!

commit 2f16cea4 (2017-10-27): Fix response requestHandle in OpenSecureChannel (fix #1276)
closed 2017-10-27 by jbvoelker