open62541/open62541 #1278: Multiple connections to server/Multithreading OS x

WoutvdBerg, 2017-10-26:
Description
===========
Is it possible to establish multiple connections to different servers. I have a class where the client is created and the data is retrieved. It works perfect but when I add an extra instance of the class and connect it, I get different tcp warnings, tcpinternalerror, badinvalidargument ( I think because the information gets switched and the response is not right for the request.) 

I've looked into the multithreading but I'm on OS X and since multithreading is experimental I would like to know if there is another option?

Using amalgamated open62541

Background Information / Reproduction Steps
===========================================
Create a client class with open62541 and make 2 instances of that class with both another ip and port combination to connect to different servers. Then read/write both server simultaneously.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.2
 - [ ] Other OPC UA SDKs used (client or server): Prosys simulation Server, UaCpp server
 - [ ] Operating system: OS x
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-10-27:
Hello @WoutvdBerg ,

you can just switch to the current 0.2 branch and it should work right away.
The difference is that thread-local storage is always enabled, regardless of whether full-multithreading support is enabled.

If you can wait a few days, we are about to package a v0.2.1 maintenance release that will contain the change.

closed 2017-10-27 by jpfr

WoutvdBerg, 2017-10-27:
@jpfr have you tested this with OS X? I have problems activating thread-local storage on my mac, the clang that comes with the command line tools doesn't support thread local storage. I've tried installing llvm clang but can't get netbeans to use this.

jpfr, 2017-10-27:
We build every commit on osx.
See here for the travis build matrix: https://github.com/open62541/open62541/blob/0.2/.travis.yml

WoutvdBerg, 2017-11-02:
@jpfr, I can't figure out which compiler is used for mac os x.

I have tried multiple compilers but can't find a compiler that supports Thread Local Storage on os X. 
To test other issues I need the thread local storage fixed otherwise I'm limited to the old v0.2 without TLS.

Stasik0, 2017-11-02:
It uses gcc

Does this help?
https://wiki.freitagsrunde.org/GCC_unter_Mac_OS

Pro, 2017-11-02:
Travis uses clang:
https://github.com/open62541/open62541/blob/a4d0ea98ab8200a6d8d2c4394522ad243010aa48/.travis.yml#L49

Here you can see the version:
https://travis-ci.org/open62541/open62541/jobs/296226523#L65
`Apple LLVM version 7.3.0 (clang-703.0.31)`

According to the latest travis build, multithreading should at least build with that version on master

See:
https://github.com/open62541/open62541/blob/a4d0ea98ab8200a6d8d2c4394522ad243010aa48/tools/travis/travis_osx_script.sh#L39
 and
https://travis-ci.org/open62541/open62541/jobs/296226523#L3740

Stasik0, 2017-11-02:
Oh no, I read the exclude portion, sorry

WoutvdBerg, 2017-11-02:
Does a example run with the builded library, I can built the library itself(like travis does?) but when I try to compile an example it won't compile because of the thread local storage. I can check myself with Apple LLVM version 7.3.0 tomorrow.

jpfr, 2017-11-03:
Support for the (C++11) thread_local was added to XCode 8.
I don't know about the C11 keyword _Thread_local.

Anyway, PR #1300 adds support for the C++ way of defining thread-local variables.
In the worst case, you can compile the main lib as C++.

WoutvdBerg, 2017-11-03:
That means with the standard compilers from Xcode 9 it's possible to use thread local storage?

jpfr, 2017-11-03:
Yes.

WoutvdBerg, 2017-11-06:
Found out what the problem was, the project has compiler flags that OS X 10.6 has to be supported but thread local storage doesn't. From 10.7 it works, I have to find a solution for this problem until then I can't test the other open issues from me.