open62541/open62541 #1283: When monitorring several variables,how can I know which variable has been chanced?

Croton-D, 2017-10-28:
I tried client.c to monitory some variables.
1:  add monitoredItem like this:
UA_NodeId monitorThis = UA_NODEID_STRING(4, "A");
UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE,
                                             &handler_TheAnswerChanged, NULL, &monId);
UA_NodeId monitorThis = UA_NODEID_STRING(4, "B");
UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE,
                                             &handler_TheAnswerChanged, NULL, &monId);
2:change the handler_TheAnswerChanged function as:
handler_TheAnswerChanged(UA_UInt32 monId, UA_DataValue *value, void *context) {
    printf("monid is %d  and value is %f\n",monId,*(UA_Float*)value->value.data);//the variables are all floats
}
3: call the function to get whether varibles has been changed:
UA_Client_Subscriptions_manuallySendPublishRequest(client);
sleep(200);
UA_Client_Subscriptions_manuallySendPublishRequest(client);

then ,I got the message ：
monid is 1  and value is 1.500000
monid is 1  and value is 2.000000
monid is 1  and value is 2.500000

using UA_Expert,I found A was 1.5 and B was 2.0 changing to 2.5,but the messages reported the same monid.
My question is how can I know which variable has been changed when then has the same monid?

StalderT, 2017-10-29:
You can use context, like : 

```
UA_NodeId monitorThis = UA_NODEID_STRING(4, "A");
UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE,
&handler_TheAnswerChanged, (void*)1, &monId);
UA_NodeId monitorThis = UA_NODEID_STRING(4, "B");
UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE,
&handler_TheAnswerChanged, (void*)2, &monId);
```

and

```
handler_TheAnswerChanged(UA_UInt32 monId, UA_DataValue value, void* context) {
printf("monid is %d and value is %f context %d\n",monId,(UA_Float)value->value.data, (int) context);//the variables are all floats
}
```

Code should be tested!

jpfr, 2017-11-03:
Can you try with the current 0.2 branch or master branch?
I think this is no longer an issue.

Infinity95, 2017-11-20:
Can we close this issue, or does the problem still persist?

Pro, 2018-01-17:
@Croton-D I will close this issue for now. If you still have further questions feel free to reopen.

closed 2018-01-17 by Pro