open62541/open62541 #1284: Segmentation fault in client in UA_Client_Subscriptions_manuallySendPublishRequest

StalderT, 2017-10-29:
Description
===========

client crash in UA_Client_Subscriptions_manuallySendPublishRequest

Background Information / Reproduction Steps
===========================================

Modify client.c : 

line 108 :

```
    UA_Client_Subscriptions_manuallySendPublishRequest(client);
```

to

```
    while(1){
        UA_Client_Subscriptions_manuallySendPublishRequest(client);
        sleep(1);
    }
```

start bin/examples/server in a shell
start bin/examples/client in a shell

in server shell press ctrl+z
wait 20 seconds
press fg in shell

client crash. Valgring result : 
```
==8535== Stack overflow in thread #1: can't grow stack to 0xbe710000
==8535== 
==8535== Process terminating with default action of signal 11 (SIGSEGV)
==8535==  Access not within mapped region at address 0xBE710FDC
==8535== Stack overflow in thread #1: can't grow stack to 0xbe710000
==8535==    at 0x804BB45: ??? (in /data/open62541-master/bin/examples/client)
==8535==  If you believe this happened as a result of a stack
==8535==  overflow in your program's main thread (unlikely but
==8535==  possible), you can try to increase the size of the
==8535==  main thread stack using the --main-stacksize= flag.
==8535==  The main thread stack size used in this run was 8388608.
==8535== Stack overflow in thread #1: can't grow stack to 0xbe710000
--8535-- VALGRIND INTERNAL ERROR: Valgrind received a signal 11 (SIGSEGV) - exiting
--8535-- si_code=1;  Faulting address: 0xBE710FDC;  sp: 0x62c15f30

valgrind: the 'impossible' happened:
   Killed by fatal signal

host stacktrace:
==8535==    at 0x58039BAE: ??? (in /usr/lib/valgrind/memcheck-x86-linux)

sched status:
  running_tid=1

Thread 1: status = VgTs_Runnable (lwpid 8535)
Segmentation fault
```

Checklist
=========
Please provide the following information:

 - [X] open62541 Version (release number or git tag): master https://github.com/open62541/open62541/commit/afd1e35a1664901e41f7ef4f624d986739fc27af
 - [X] Other OPC UA SDKs used (client or server): open62541
 - [X] Operating system: linux
 - [X] Critical issue

StalderT, 2017-10-29:
With UA_LOGLEVEL 100 : 

After fg, the server send in loop : 
```
[10/29/2017 15:19:54.420] trace/channel	Connection 4 | SecureChannel 5 | Process a MSG
[10/29/2017 15:19:54.420] info/session	Try to use Session with token f2a8bc6b-79e3-b9bd-4084-a89a88580540 but is not found
[10/29/2017 15:19:54.420] info/channel	Connection 4 | SecureChannel 5 | Service request 473 without a valid session
[10/29/2017 15:19:54.420] debug/server	Sent ServiceFault with error code -2145058816
[10/29/2017 15:19:54.421] trace/network	Connection 4 | Activity on the socket
[10/29/2017 15:19:54.421] trace/network	Connection 4 | Received a packet.
```

until client crash.

StalderT, 2017-10-30:
In case of error the client try to disconnect

1) __UA_Client_Service call receiveServiceResponse :

https://github.com/open62541/open62541/blob/2bf6a6b80654f590e2312eb2b4fb90cd755a1576/src/client/ua_client.c#L311

2) receiveServiceResponse call UA_Client_disconnect :

https://github.com/open62541/open62541/blob/2bf6a6b80654f590e2312eb2b4fb90cd755a1576/src/client/ua_client.c#L282

3) UA_Client_disconnect call sendCloseSession :

https://github.com/open62541/open62541/blob/2bf6a6b80654f590e2312eb2b4fb90cd755a1576/src/client/ua_client_connect.c#L523

4) sendCloseSession call __UA_Client_Service : 

https://github.com/open62541/open62541/blob/2bf6a6b80654f590e2312eb2b4fb90cd755a1576/src/client/ua_client_connect.c#L498

And now client jump to 1) and has a recursive loop until crashing (stack overflow)

jpfr, 2017-10-31:
I merged the PR. Thanks for that!
Is it possible to implement the sequence leading to the crash in a unit test?
That protects us against regressions in the future.

StalderT, 2017-10-31:
I don't know how we can simulate a network link failure. Perhaps/Probably just adding a long sleep somewhere in server ?

StalderT, 2017-10-31:
something like that : 

[client_test_network_failure.txt](https://github.com/open62541/open62541/files/1430824/client_test_network_failure.txt)
[server_test_network_failure.txt](https://github.com/open62541/open62541/files/1430825/server_test_network_failure.txt)

client_test_network_failure fail before https://github.com/open62541/open62541/issues/1284 and works after

I do not know how to modify these examples in unit test.

jpfr, 2017-10-31:
Thanks you. We can construct a unit test out of this.

MPJansen, 2017-11-01:
I ran into this yesterday when I specified an unreasonably small publishing interval for my monitored items.
The server seemed to lock up and generate stackoverflows and tried reaching memory that was denied access.

I can confirm that updating to head and picking a more reasonable interval fixes this, thanks!

jpfr, 2017-11-03:
Closing this. A pointer for the unit test was left in #1296,

closed 2017-11-03 by jpfr