open62541/open62541 #1285: Segmentation fault in manually compiled server.c

StalderT, 2017-10-30:
Description
===========
manually compiled server.c crash

Background Information / Reproduction Steps
===========================================
build open62541 with ccmake . 
```
 BUILD_SHARED_LIBS                OFF                                                                                                                                                                                                      
 CMAKE_BUILD_TYPE                 Release                                                                                                                                                                                                  
 CMAKE_INSTALL_PREFIX             /usr/local                                                                                                                                                                                               
 LIB_INSTALL_DIR                  /usr/lib                                                                                                                                                                                                 
 UA_BUILD_EXAMPLES                ON                                                                                                                                                                                                       
 UA_BUILD_UNIT_TESTS              OFF                                                                                                                                                                                                      
 UA_ENABLE_AMALGAMATION           ON                                                                                                                                                                                                       
 UA_ENABLE_COVERAGE               OFF                                                                                                                                                                                                      
 UA_ENABLE_DISCOVERY              OFF                                                                                                                                                                                                      
 UA_ENABLE_DISCOVERY_MULTICAST    OFF                                                                                                                                                                                                      
 UA_ENABLE_FULL_NS0               OFF                                                                                                                                                                                                      
 UA_ENABLE_METHODCALLS            ON                                                                                                                                                                                                       
 UA_ENABLE_NODEMANAGEMENT         OFF                                                                                                                                                                                                      
 UA_ENABLE_SUBSCRIPTIONS          ON                                                                                                                                                                                                       
 UA_LOGLEVEL                      100                                                                                                                                                                                                      
 UA_VXWORKS_WRS_KERNEL            OFF
```

bin/examples/server works

now compile server with : 

```
gcc examples/server.c open62541.c -I./ -o server
```

server crash with : 

```
==4336== Memcheck, a memory error detector
==4336== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==4336== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==4336== Command: ./server
==4336== 
==4336== Conditional jump or move depends on uninitialised value(s)
==4336==    at 0x402C33A: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==4336==    by 0x804F939: UA_Array_delete (in /data/open62541-master/server)
==4336==    by 0x804F6EB: deleteMembers_noInit (in /data/open62541-master/server)
==4336==    by 0x804F71A: UA_deleteMembers (in /data/open62541-master/server)
==4336==    by 0x804C183: UA_String_deleteMembers (in /data/open62541-master/server)
==4336==    by 0x8058A9F: UA_Server_initNS0 (in /data/open62541-master/server)
==4336==    by 0x8056D46: UA_Server_new (in /data/open62541-master/server)
==4336==    by 0x80495A1: main (in /data/open62541-master/server)
==4336== 
==4336== Invalid free() / delete / delete[] / realloc()
==4336==    at 0x402C377: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==4336==    by 0x804F939: UA_Array_delete (in /data/open62541-master/server)
==4336==    by 0x804F6EB: deleteMembers_noInit (in /data/open62541-master/server)
==4336==    by 0x804F71A: UA_deleteMembers (in /data/open62541-master/server)
==4336==    by 0x804C183: UA_String_deleteMembers (in /data/open62541-master/server)
==4336==    by 0x8058A9F: UA_Server_initNS0 (in /data/open62541-master/server)
==4336==    by 0x8056D46: UA_Server_new (in /data/open62541-master/server)
==4336==    by 0x80495A1: main (in /data/open62541-master/server)
==4336==  Address 0x804862c is in a r-x mapped file /data/open62541-master/server segment
==4336== 

```

Checklist
=========
Please provide the following information:

 - [X] open62541 Version (release number or git tag): master https://github.com/open62541/open62541/commit/2bf6a6b80654f590e2312eb2b4fb90cd755a1576
 - [X] Operating system: linux
 - [X] Critical issue

jpfr, 2017-10-30:
The fix looks good. Can you make a PR?
Thanks.

jpfr, 2017-10-30:
Fixed via #1286.

closed 2017-10-30 by jpfr